esp8266/Arduino #1029: Exception(28) on client.connect() when connecting to api.parse.com using parse-sdk

jcard0na, 2015-11-17:
Hi,

I'm trying to port parse.com's sdk to Arduino/esp8266.  I was able to connect over SSL to their api (api.parse.com) successfully.  After that, I modified their SDK to support esp8266 but I am getting an Exception(28).  The code is available here:  https://github.com/jcard0na/parse-sdk-for-esp8266

```
connecting to blah
.......
WiFi connected
IP address: 
192.168.1.127
begin(<--key omitted -->, <--key omitted -->)
Parse hello world example started
basic object operation test
create...
create.send()...
sendRequest("POST", "/1/classes/Moobean", "{"toBlah":false,"copied":false}", "")

Exception (28):
epc1=0x402027cf epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: sys 
sp: 3fff04d0 end: 3fffffb0 offset: 01a0
```
I disassembled the code to find the offending instruction.  It is:

```
402027c6:       fff405          call0   40202708 <_ZN16ESP8266WiFiClass10hostByNameEPKcR9IPAddress>
402027c9:       429c            beqz.n  a2, 402027e1 <_ZN10WiFiClient7connectEPKct+0x39>
402027cb:       0c28            l32i.n  a2, a12, 0
402027cd:       013d            mov.n   a3, a1
402027cf:       6258            l32i.n  a5, a2, 24
```

It is worth mentioning that I was able to connect successfully to that same URL using the HTTPSRequest example sketch.

I read somewhere that Exception(28) might be caused by power issues, but using a bench power supply to power up my board makes no difference.

I am using the latest code from master from this repository (checked out today).
My board is Adafruit's Huzzah.

Can anyone help out?

Thanks,

Javier

igrr, 2015-11-17:
Hi Javier, 
I was able to get your ParseHelloWorld example to work.

Inside ParseHelloWorld example I noticed that in `basicObjectTest` function you create a total of 4 `ParseResponse` object, each is >1024 bytes in size. All these objects are allocated on the stack, and we only have 4k of stack. The exception you have encountered is likely caused by stack running into heap and damaging some objects there (or vice versa). When I have surrounded each response in its own `{}` scope, all went fine. I saw objects being created in Parse dashboard.

The only (minor) change to the library itself was to add `yield();` in ParseClient.cpp:208, to give some time for TCP stack to do it's business between connection attempts. You could replace that with `delay(some_timeout_in_ms);` if it makes more sense.

Make sure you rename `parse_hello_world` directory under examples to `ParseHelloWorld`, otherwise the IDE won't see this example (directory and sketch name should be the same).

Also note that I have pushed some changes to TLS just two hours before you opened this issue. Make sure you grab at least a2789df. If you are using git version inside your `hardware` directory, check that you don't have boards manager package installed, otherwise IDE will use it instead of the contents of `hardware` directory.

jcard0na, 2015-11-17:
Oh, wow! That's great news!  I will test it tomorrow and report back.

Thanks!

jcard0na, 2015-11-19:
I have re-tested with the two changes you suggested (reduce stack usage + yield() while connecting) and it now works well.

Thanks!

N0TB0T, 2015-11-21:
@jcard0na Would you mind updating your example, so it works out of the box with the changes @igrr suggested? ;)

And to you both: I could not be more thankful. Waited so desperately for https and parse to happen!

N0TB0T, 2015-11-21:
I'm just watching data sent to parse now for minutes. Such a beauty. =)

igrr, 2015-11-21:
Good to know it works!

closed 2015-11-21 by igrr

N0TB0T, 2015-11-21:
Oh, and @jcard0na - your example suggests to use the REST API KEY, but then sends it as client key. Took me another hour to find that out.

igrr, 2015-11-21:
Oh yeah, that key naming got me stumbled as well.

jcard0na, 2015-11-21:
Hi guys,

Thanks for the feedback.  Yes, I was not sure which of those two keys I had
to use and experimented with both.  Unfortunately the comment called for
the wrong one.

My changes have since been merged upstream here:
https://github.com/ParsePlatform/Parse-SDK-Arduino

I will fix the example and the key naming there.

Best,

Javier

atulpriyadarshi21992, 2016-01-16:
Hello Everyone,
I am also using the same library to communicate between parse.com and ESP-07

When I run the ParseHelloWorld.ino example in the library, it runs fine till a while then WDT reset kick in and the ESP resets.
This happens after 10-16 time the test is succesfully run.
The reset key work is

ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

This generally occurs in GET/UPDATE function, although sometimes it also occurs in PUT segment.

Hope you all can help