esp8266/Arduino #1031: Arduino 1.66 Windows 10 - no ESP

scargill, 2015-11-17:
Hi there

Arduino 1.66 - fresh clean version on a pretty fresh Windows 10 update... when I go to preferences and paste the link...

http://arduino.esp8266.com/stable/package_esp8266com_index.json

Nothing happens. No new boards... am I alone?

igrr, 2015-11-17:
Hi Pete,
you're not alone, there are a few issues related to 1.6.6 ([list](https://github.com/esp8266/Arduino/labels/arduino%201.6.6)). At the moment you should probably try 1.6.5.

scargill, 2015-11-17:
Ok, downloading now... thanks Ivan! One massive download on the way...

scargill, 2015-11-17:
Ok, one more question. I SCRAPPED all Arduino and started from empty - I install 1.6.5  - installation ok - I put the same json in - then went to boards - and added the ESP stuff in - 

So the ESP boards are available... YES... but no examples - no libraries. I tried copying the whole lot from GIT to:

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266

Sure enough - restart Arduino and the examples ran - but would not compile (generic ESP8266) -  I've had all of this running on other machines in the past...

Any ideas where I'm going wrong here. Clean Arduino installation....

igrr, 2015-11-17:
Could you please post compiler errors you are seeing?

per1234, 2015-11-27:
>Nothing happens. No new boards... am I alone?

You need to open Boards Manager twice before the board for a newly added Boards Manager URL will appear in Boards Manager. This is a bug in the Arduino IDE https://github.com/arduino/Arduino/issues/3795 that has been fixed but is only available in the hourly build.

closed 2016-02-29 by igrr