esp8266/Arduino #1032: WPA2-enterprise + PEAP

kadu, 2015-11-17:
Can ESP connect to WPA2-enterprise + PEAP networks ? If yes, how to do that ?

bakercp, 2015-11-18:
There is a longer discussion of this topic on the esp8266 forum http://www.esp8266.com/viewtopic.php?f=6&t=1243&start=15.

igrr, 2015-11-30:
WPA2-enterprise is supported in SDK 1.5, so linking this to #1102.

joostd, 2015-12-12:
I suspect only eap-tls is supported.

See http://bbs.espressif.com/download/file.php?id=988. The only reference to wpa2-enterprise is in
wifi_station_set_cert_key 

Also:

$ ar -t esp_iot_sdk_v1/esp_iot_sdk_v1.5.0/lib/libwpa2.a 
/
//
asn1.o/
base64.o/
bignum.o/
eap_common.o/
eap.o/
/0
eap_tls.o/
ext_password.o/
pkcs1.o/
pkcs5.o/
pkcs8.o/
rsa.o/
tls_internal.o/
tlsv1_client.o/
/18
/39
tlsv1_common.o/
tlsv1_cred.o/
tlsv1_record.o/
tlsv1_server.o/
/61
/82
x509v3.o/

JeroenBeemster, 2015-12-12:
HI guys. Finally WPA2 Enterprise is supported. But I cannot find any example. Actually I saw that only 1 person got it working. 
 My question is: I need to fill in a username (identity) and a password for the WPA2 and no private key or something like that. Where to fill in the "username" ? Or is this not supported yet. I know this is not an arduino question. But anyway here are the specialists :)

igrr, 2015-12-18:
I have not yet added `-lwpa2` to linker flag because I wasn't able to get WPA2-Enterprise to work.
If anyone wants to try, feel free to add this flag [here](https://github.com/esp8266/Arduino/blob/master/platform.txt#L27)

pauldekkers, 2015-12-21:
@igrr shouldn't that be line 33? Either way that doesn't work for me, I included

#include <user_interface.h>
...
wifi_station_clear_cert_key();

... in my code to see if that would build, but it doesn't:

sketch_dec20a.ino:5: undefined reference to `wifi_station_clear_cert_key()'
collect2: error: ld returned 1 exit status

But I'm not sure this is the right way to call those functions.

igrr, 2015-12-21:
You need to wrap that include with `extern C`:
```c++
extern C {
#include "user_interface.h"
}
```

pauldekkers, 2015-12-21:
@igrr extern "C" does get me further, except for this:

/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lib/libcrypto.a(aes-internal-dec.o):(.irom0.text+0x4): undefined reference to `Td0'
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lib/libcrypto.a(aes-internal-dec.o): In function `aes_decrypt_init':
(.irom0.text+0x134): undefined reference to `Td4s'
collect2: error: ld returned 1 exit status

I hope you know what's missing :-)

JeroenBeemster, 2015-12-21:
I also added the  -lwpa2 parameter. 
I have exact the same error as Paul.
Same result when I tried this new function: wifi_station_clear_cert_key();

igrr, 2015-12-22:
Should be fixed by 70cf3c3

JeroenBeemster, 2015-12-22:
Yes, now it compiles. Thanks Igrr, Back to my first question. Where to put the username?
How does it work? I am missing one part of the puzzle. And I do not know which

pauldekkers, 2015-12-22:
@igrr Great, that compiles and I've actually got it to authenticate. But: with the wrong User-Name, that seems to be set to "tianhao" and not taken from the certificate CN. Maybe it uses the Microsoft UPN attribute or a subjectAltName; I'll try to experiment with that, but suggestions are welcome.

DavidXanatos, 2016-01-03:
I also need to connect to a WPA2 Enterprise secured network, it requirers user authentication that is Username + Password, no certificate.
The server certificate is a self signed one.

Did anyone got this to work? And if so is there a guide + example, I guess I will need to update the firmware on the module so a guide for that would also be appreciated.

pauldekkers, 2016-01-04:
So far only EAP-TLS is supported; so client-certificate based and not username/password. This works fine for me, but if you have no client-certificate and no control over the RADIUS server, I don't think you can do a lot (but wait for a series of other EAP-mechanisms to be supported).

DavidXanatos, 2016-01-04:
I have full control over the RADIUS server just never configured anything else than user name based.
If I have to add certificate based (if that can coexist) than its doable.
Still how do I do that on the arduino side?

joostd, 2016-02-15:
You can use something like https://github.com/joostd/esp8266-eduroam/blob/master/Arduino/eduroam/eduroam.ino. At the moment, it needs a patched esp8266 Arduino core (https://github.com/esp8266/Arduino/pull/1633)

closed 2016-02-27 by igrr

someburner, 2016-04-17:
FYI Espressif just released 1.5.3 which adds username/password support.

joostd, 2016-04-19:
Alas, all that 1.5.3 adds are the missing prototypes in `include/user_interface.h` (`wifi_station_set_username` and `wifi_station_clear_username`). These are not used for username/password authentication, but for setting/clearing the username used in EAP. As such, EAP-TLS (i.e. using client certificates) is still the only supported EAP method.
(In fact `lib/libwpa2.a` has not changed since version 1.5.2).

It does mean that [PR #1633](https://github.com/esp8266/Arduino/pull/1633) is no longer necessary once the SDK is upgraded to 1.5.3.

someburner, 2016-04-19:
@joostd  Yeah I didn't notice that at first.. got a bit excited. I haven't been able to get PEAP going at my workplace.

Matthew-Bradley, 2016-08-13:
I'm very interested. Can someone give a quick run done of the step by step for what almost "works" right now so I could give it a try? I take it it's only certificate based and not identity/password based that works (would be nice, but ok).

bloons3, 2016-08-22:
I'm interested in seeing the arduino implementation for username/password on 801.2x

Matthew-Bradley, 2016-10-28:
What's the current status of this?

joostd, 2016-10-28:
With the switch to V2.0 of the espressif sdk, EAP-TTLS and PEAP should now work as well.

bospre, 2016-10-29:
Hi @ all,
but where is an example on how to use it?
I could not find one.
Does anybody has an example for WPA2 with PEAP?

joostd, 2016-10-29:
If you look at the (eap-tls) code at
https://github.com/joostd/esp8266-eduroam/blob/master/Arduino/eduroam/eduroam.ino

you should be able to use PEAP instead of EAP-TLS by replacing a few lines, most notably replace calls like
wifi_station_set_cert_key 
with calls to
wifi_station_set_wpa2_enterprise_auth
wifi_station_set_enterprise_username
wifi_station_set_enterprise_password

See also:
https://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf

Hope this helps…

cmfcmf, 2016-10-30:
It appears to me that this framework doesn't yet use version 2 of the SDK, but there is an open pull request: https://github.com/esp8266/Arduino/pull/2440.

gorghino, 2016-10-30:
Hi @cmfcmf, this's a branch of Esp8266/Arduino that it uses SDK2 BUT WPA2-PEAP doesn't seem to work #2595 
If you want to test it you can follow these directions: https://github.com/esp8266/Arduino/issues/2304#issuecomment-245849344

I tried to use ESP openSDK too, that it uses SDK2 but it doesn't seem to work neither.
Check here: https://github.com/pfalcon/esp-open-sdk/issues/220

bospre, 2016-10-30:
Hi jjoostd,
I get: 'wifi_station_set_enterprise_password' was not declared in this scope.
I only included <ESP8266WiFi.h>
If I try to include "wpa2_enterprise.h"
I get: wpa2_enterprise.h: No such file or directory

So the question is: Do I need to include wpa2_enterprise.h?
And where is the file located?

bospre, 2016-10-30:
Sorry, there was a typo.
It should  be
I only included <ESP8266WiFi.h>

bospre, 2016-10-30:
Sorry again
I pasted the name  but it disappeared after sending.
I meant I only included ESP8266WiFi.h

gorghino, 2016-10-30:
@bospre You need to use another branch of this project.
Follow these directions: https://github.com/esp8266/Arduino/issues/2304#issuecomment-245849344

bospre, 2016-10-31:
Thank you for this hint.
But how is this done with WIN-OS?
I think this only works for Linux?
Can't the files be downloaded manually?

gorghino, 2016-10-31:
I think you can clone a branch in Windows too (http://stackoverflow.com/questions/1778088/how-to-clone-a-single-branch-in-git) but you have to change the Arduino linux folder with yours.

noerw, 2016-11-07:
how to connect to PEAP (source: [sdk 2.0 documentation](https://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf), page 133):

1. Call `wifi_station_set_config` to set the configuration of target AP.
2. Call `wifi_station_set_wpa2_enterprise_auth(1);` to enable WPA2_Enterprise authentication.
    - For EAP-TLS authentication, call `wifi_station_set_enterprise_cert_key` to set certificate and > private key.
    `wifi_station_set_enterprise_username` is an optional choice, it can be called to set user name.
    - For EAP-TTLS or EAP-PEAP authentication, call `wifi_station_set_enterprise_username` and > `wifi_station_set_enterprise_password` to set user name and password.
    `wifi_station_set_enterprise_ca_cert` is an optional choice, it can be called to set root certificate.
3. Call `wifi_station_connect` to connect to target AP.
4. After being connected to an AP, or failing to connect to AP and on stopped retries, please call the corresponding `wifi_station_clear_enterprise_XXX` APIs to release the resources.

gorghino, 2016-11-07:
@noerw That is what Gidrix (and me too) tried here: https://github.com/esp8266/Arduino/issues/2595 
Something doesn't work

bospre, 2016-11-11:
Do you have access to the radius server?
What does the logfile say?
I did not manage the installation of the sdk 2.0 with arduino.
Maybe if somebody can send me the complete package ready to use.
I found it lying under 
C:\Users\*replaceyourusername*\AppData\Roaming\Arduino15\packages\esp8266\

I am running my own radius server, so I have access to the logfiles and maybe I can find out, why it does not work.

spm5065, 2016-11-15:
@noerw I have done what you said, and I always get wrong password returned from `wifi_station_get_connect_status`. I know that the username and password supplied to  `wifi_station_set_enterprise_password` and `wifi_station_set_enterprise_username` are correct, so I believe the problem to be in the `wifi_station_set_config`. Do I need to set anything other than the SSID in the station config?

gorghino, 2016-11-16:
Espressif wrote me yesterday.
They said:

> Hi,
> Please see sample code below.
> And we are woking on an introduction about WPA2 enterprise. Please wait a couple of days, and we will release on our website.
> 
> ```
> void ICACHE_FLASH_ATTR
> user_init(void)
> {
> int ret;
> 
> struct station_config wifi_config;
> os_printf("User Init\n");
> os_memset(&wifi_config,0,sizeof(wifi_config));
> os_strcpy(wifi_config.ssid, "TP-LINK_92FC", os_strlen("TP-LINK_92FC"));
> os_strcpy(wifi_config.password, "12345678", os_strlen("12345678"));
> wifi_set_opmode(STATION_MODE);
> wifi_station_set_config(&wifi_config);
> 
> wifi_station_set_wpa2_enterprise_auth(1);
> os_printf("wpa2 set CA Certificate\n");
> ret = wifi_station_set_enterprise_ca_cert(ca, os_strlen(ca) + 1);
> wifi_station_set_enterprise_username("espressif", os_strlen("espressif"));
> wifi_station_set_enterprise_password("test11", os_strlen("test11"));
> 
> wifi_station_disconnect();
> wifi_station_connect();
> 
> }
> ```

rbmb, 2016-11-18:
Hi gorghino,
I'm working on the subjects since 2 days, and I think I have found the cause.
In order to "make it" working, so I setup the following system : 
- A virutal machine to have a RADIUS server : FreeRADIUS (downloaded from RADIUSDesk)
- A small Wifi routeur that is configured to WPA2 Entreprise and pointing to the virtual machine which is running the RADIUS server.
- Wireshark packet analyzer.
- Arduino 1.6.9 , with esp8266 arduino SDK 2.0.0 Running on windows. (It was a bit difficult but managed to make it compile and work)  a call to get_sdk_version shows  : 2.0.0 (5a875ba)

So I tried sketchs from from Gidrix,  Joostd (modified according his suggestion), the one that Espressif sent you, and many other combination derived from these with different call order for initialization.

Of course none of them worked.
By looking at the packets it's obvious : the esp always sends the same identifier, which is anonymous@espressif.com,  instead of the one defined by wifi_station_set_enterprise_username
[image]

This string is present in libwpa2.a file, generated from eap.c.

If you change the string in this file, and recompile the sketch, the esp will send the identifier that you put in it.
[image]

Is it a default value ?  The wifi_station_set_enterprise_username  method is expected to change it, but it doesn't.
Have we access to the  wifi_station_set_enterprise_username  source code ? What is broken in it ? 
or the modified eap.c (derived from https://w1.fi/wpa_supplicant/devel/eap_8c.html) file from espressif ? 

Where is the default value of the password in this file ? maybe we can put the identifier in it and try if it is connecting.

Please post the sketch you want me to try, so I can look if it changes the identifier and send you the result.
Best regards,
René

alviso, 2016-11-19:
Hi @rbmb 
I was also working on this trying to connect to UPC Wi-Free. Edited libwpa2.a and replaced with my username but it didn't seem to help either but thanks for the tip. Hopefully soon someone makes a breakthru.
Peter

rbmb, 2016-11-20:
@Alviso,
The workaround with libwpa2.a is only half-way at the moment. We can only set the username, but the password is still missing.

- Either we know where it is in this file and know how it's encoded, so we can set it at compile time

- Either somebody from Espressif (or else) should tell the default password, but in this scenario we should have some control upon the RADIUS server to put the password that we want (the default one).

I don't know why set_username function won't work with Arduino.
SetAuthentification function is working.
 According to Gidrix, it's working OK with NONOS SDK2.0.0. https://github.com/esp8266/Arduino/issues/2595

The file libwpa2.a is exactly the same,  the header file wpa2_entreprise.h is also the same except the comments at the beginning between Arduino and NONOS SDK. 
How can we progress in this issue ? Where should we look ?
Can it be the linking order in platform.txt (compiler.c.elf.libs)?

JimmyTai, 2016-12-06:
@rbmb Thanks for your nice explanation for why WPA2 Enterprise is not working in Arduino now. Do you have any progress for solving this identifier and password problem or any news from Espressif? Thank you again!

rbmb, 2016-12-12:
Hi Jimmy,
Unfortunately, i didn't solved the issue. I don't have any contacts with espressif nor with igor. Hope someone can route this findings to them.
Meanwhile, i was forced to change the board for this project due to this issue.
I ended using wired Leonardo eth. from.arduino.org. The ethernet connection isnot reliable (due to hardware i think) and 97% of program flash is used.

joostd, 2016-12-12:
Actually, `anonymous@espressif.com` is the "outer identity" which is used for routing RADIUS messages, not for user authentication, and it is not supposed to be set using `wifi_station_set_enterprise_username`. That function is used for setting the "inner identity" used in PEAP or TTLS. The reason why authentication fails (at least in my case) is that there is an issue with the method used for user authentication (usually MSCHAPv2).

pepe79, 2016-12-18:
I finally got an IP with WPA2 PEAP using the sdk 2.0.0 branch: see my comment at  https://github.com/esp8266/Arduino/issues/2595#issuecomment-267809669

jtuttas, 2017-01-17:
Hello,
just searched a while and still returned to this post, so I have to ask here !

I Try to connect a ESP8266-E12 to a WPA2-Enterprise Network, that uses PEAP und MSCHAPv2 for authentication with a username and a password, but i can'nt get it working.

So any one here has a **working example** for this configuration? Is is possible?

best regards 

Jörg Tuttas

ninjabe86, 2017-01-18:
Hi @jtuttas 

I got my esp8266 connected to the eduroam network which uses (AFAIK) PEAP MSCHAPv2.
It is sending data to a thingspeak graph and works well.

To do this you will need to 

1. Follow @igrr instructions on how to install the update sdk 2.0 version into your arduino folder.

2. Go to the #2595 comment thread and copy the code at the top into a sketch. Be sure you make the changes to the code as @pepe79 has pointed out.

3. Complie the code to ensure it works.

4. Edit the binary file lipwap2.a using a binary editor. hex edit for windows or bless for linux works well
 .When editing the binary file, you are looking for the string with "anonymous@espressif.com"
 Once you find that string in your binary editor window, you must edit it to match your username for the network. NOTE: If you edit the file be sure to to fill any renaming space with "."s to ensure the file stays the same size . 
anonymous@expressif.com to

.....your@your.com...... 

you need to pad out the spaces left. in this case anonymous is 9 letters and your is 4, therefore you need 5 dots to make up the difference and stop the binary file becoming corrupted.

5. Once you have edited and saved the binary file you can now complie and upload your code to the esp8266 and it should work. Below is the code I used to get my readings from a pot and upload to things speak (my credentials have obviously been removed)

 `#include <Arduino.h>
#include <ESP8266WiFi.h>
#ifdef ESP8266
extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}
#endif

static u8 ent_username[] = "yourusername"; // same as the mdoification to lipwpa2.a file
static u8 ent_password[] = "your password";
const char* host = "api.thingspeak.com";

void setup(){
  pinMode(LED_BUILTIN, OUTPUT);
  char a[100];
  ip_info info;
  wifi_get_ip_info(0, &info);

  Serial.begin(115200);

  wifi_station_disconnect();
  wifi_set_opmode(STATION_MODE);

  char ssid[32] = "yourssid";
  char password[64] = {0x00};
  struct station_config stationConf;
  stationConf.bssid_set = 0;  //need not check MAC address of AP
  memcpy(&stationConf.ssid, ssid, 32);
  memcpy(&stationConf.password, password, 64);

  if(!wifi_station_set_config(&stationConf)){
    Serial.print("\r\nset config fail\r\n");
  }

  // switch to WPA2 Enterprise 
  wifi_station_set_wpa2_enterprise_auth(1); 

  if(wifi_station_set_enterprise_username (ent_username, strlen((char*)ent_username))){
    Serial.print("\r\nusername set fail\r\n");
  }
  if(wifi_station_set_enterprise_password (ent_password, strlen((char*)ent_password))){
    Serial.print("\r\npassword set fail\r\n");
  }

  if(!wifi_station_connect()){
    Serial.print("\r\nconnect fail\r\n");
  }

  Serial.print("\r\ntrying to connect...");

  while(info.ip.addr == 0){
    ESP.wdtFeed();
    Serial.print(".");
    delay(1000);
    wifi_get_ip_info(0, &info);
  }

  sprintf(a, "%"PRIu32,info.ip.addr);
  Serial.print("\r\nip addr: ");
  Serial.print(a);
  Serial.print("\r\n");
}

void loop()
{ 
  WiFiClient client;
  const int httpPort = 80;

  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  else
  {
    Serial.println("connected !");
  }

  //client.print("GET /update?key=82RS9VJ7YHQGMWT1&field1=1000\n");

  char buffer[200];
  int adcValue = analogRead(A0);
  Serial.println(adcValue);
  sprintf(buffer, "GET /update?key=IYFDMJ5JMSCC8NP4&field1=%d\n", adcValue);
  client.print(buffer);

  while (client.available()) {
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }

  client.stop();
  Serial.println("closing connection");
  digitalWrite(LED_BUILTIN, LOW);  // Turn the LED off by making the voltage HIGH
  delay(2000);
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);

delay(60000);
}
`

I hope this helps you or anyone else who was trying to get it connected.

This is my first ever post on Github so apologies if I have not followed certain conventions on this forum/thread in terms of posting/referencing

cheers

ninjabe86 :)

jtuttas, 2017-01-19:
Hi @ninjabe86 ,
wow, this a a real good conclusion of all post I have read in the past about ESP8266 and WPA2 Enterprise connections, Congratulation ! I will try all the steps within the next week and give you and the community a feedback. I guess my main falt was step4 (edit the   lipwap2.a).

best regards !

jtuttas

bospre, 2017-01-20:
Hello
I tried what you wrote above, but it did not work for me.
At first I created at the RADIUS-Server a user "anonymous@espressif.com" with password "test".
But the logfile of the RADIUS only showed login errors.
Then I tried to change libwpa2.a as you proposed, but that did not work either.
I am still hoping that sometimes there will be an update of the sdk.

ninjabe86, 2017-01-26:
@jtuttas I made a typo and corrected for it just there. be sure to edit the **libwpa2.a** file not lipwap2.a(i dont think this file even exists but just to be sure)

This is located (in my case) as follows arduino-1.6.5-r5/hardware/esp8266com/esp8266/tools/sdk/lib

@bospre  if you have updated the sdk to 2.0.0 as above and have edited the binary file without corrupting it then I'm afraid I can't help you much more. I am fairly new to this myself. 

As you said, hopefully this repo gets updated soon.

saichintha, 2017-03-09:
Hey guys,

I install the sdk 2.0 using git clone into Documents/Arduino/esp8266 and also think I modified the binary file successfully. 

However, I created a new sketch file in the Arduino IDE and saved it in the Documents/Arduino/esp8266sketch and now it can't gain access to the `wpa2_enterprise.h` in the other folder. How exactly do  I include the whole sdk in my project so my code has access to all the functions in the sdk?

Here is the error I'm getting.
```
/Users/saichintha/Documents/Arduino/esp8266epa2e/esp8266epa2e.ino:6:29: fatal error: wpa2_enterprise.h: No such file or directory
 #include "wpa2_enterprise.h"
                             ^
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.
```

adriencapaine, 2017-03-09:
I managed to get a working configuration different from @ninjabe86 as I tried to modify libwpa2.a file unsuccessfully...

Here is my configuration

### Global configuration

- A BELKIN Access point with WPA2 Enterprise support
- An ESP8266 nodeMCU v1.0
- Arduino v1.6.8 intalled with SDK2.0 integration as described by @igrr .
Shortly, in folder _c:\users\USERNAME\AppData\Local\Arduino15\packages_ (in Windows, sorry about that, check you own _packages_ repository on your OS by downloading a random Board driver with the Arduino Board Manager. It should be somewhere!)
```bash
mkdir esp8266com
cd esp8266com
git clone -b update_sdk_2.0.0 https://github.com/esp8266/Arduino.git esp8266
cd esp8266\tools
python get.py
```
- A Virtual Machine with RADIUS + DHCP servers

### Network configuration

VM IP address is 192.168.10.254, AP one is 192.168.10.2.
Both addresses are static and DHCP IP range is 192.168.10.10 to 192.168.10.150 with mask 255.255.255.0.
In the AP web server, I configured the WPA2 Enterprise security to point to the VM IP address with correct RADIUS port and you I also checked it's SSID.

### RADIUS configuration

I configured my RADIUS server for EAP - PEAP MS-CHAPV2 authentification method (in _eap.conf_), then added the client IP (it's the AP) in _clients.conf_ and added a user in _users_ with login: "user1" and pass: "secret".
Files are at the end of this post.

### Arduino code

PS: I did not modify the libwpa2.a file (in esp8266/tools/sdk directory)

```c
/**
 *  Brief:
 *    This program connects to a remote WiFi Access Point with
 * WPA2 Enterprise security
 */
/*******************************************************************/
#include <ESP8266WiFi.h>
/*******************************************************************/
extern "C" {
  #include "user_interface.h"
  #include "wpa2_enterprise.h"
}
/*******************************************************************/
#define SERIAL_BAUD_RATE      57600
#define STARTUP_DELAY_MS      1000
/*******************************************************************/
// SSID to connect to
static const char* ssid = "MY_SSID";
// Username for authentification
static const char* username = "user1";
// Password for authentification
static const char* password = "secret";
/*******************************************************************/
void setup() {
  Serial.begin(SERIAL_BAUD_RATE);
  delay(STARTUP_DELAY_MS);

  // Setting ESP into STATION mode only (no AP mode or dual mode)
  wifi_set_opmode(STATION_MODE);

  struct station_config wifi_config;

  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);

  wifi_station_set_config(&wifi_config);

  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();
  
  wifi_station_set_wpa2_enterprise_auth(1);
  wifi_station_set_enterprise_username((uint8*)username, strlen(username));
  wifi_station_set_enterprise_password((uint8*)password, strlen(password));

  wifi_station_connect();

  // Wait for connection AND IP address from DHCP
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  // Now we are connected
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}
/*******************************************************************/
void loop() {
  delay(5000);
}
/*******************************************************************/
```

### Wireshark traces

I filtered incoming RADIUS packets on the VM and saved them for different configurations.
1. ESP8266 working configuration
2. My Smartphone with login: "user1" and password: "secret"
3. My Smartphone with login: "user1", password: "secret" and outer identity: "anonymous@espressif.com"

We can see that first and third RADIUS discussion are very similar.

[traces.zip](https://github.com/esp8266/Arduino/files/830406/traces.zip)

I also attach my radius configuration files - comments are in french as I got the VM from a french colleague :)

[radius.zip](https://github.com/esp8266/Arduino/files/830414/radius.zip)

Feel free to follow those steps and tell if it is reproducible.

PS: I also tried a program in which I cleared all data with certificates, password, new_password, username methods (cf: https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf - section 3.14) before setting them and connecting to the AP but it did not work. Furthermore, I could not tell the difference between set_password and set_new_password functions. In SDK guide, one is supposed to be used for EAP-PEAP authentification and the other one for MSCHAPV2. Isn't it the same?

@saichintha : I guess you have got to wrap your include with extern "C". If not working, you may have problems with your SDK2.0 integration.
```c
extern "C" {
 #include "wpa2_enterprise.h";
}
```

hholi, 2017-03-10:
Using the code from @adriencapaine and modifying the libwpa2.a as @ninjabe86 did I managed to make a connection with a network secured with WPA EAP. I used the vi editor to edit the libwpa2.a file.

cbrum11, 2017-03-21:
@ninjabe86 

I have tried diligently to follow your very detailed procedure but have run into a problem at step 3.  When I go to verify the code I get the following error message via the Arduino IDE.

**"Arduino: 1.6.9 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"**

**exec: "C:\\Program Files (x86)\\Arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling for board WeMos D1 R2 & mini.**

I have read all the subsequent threads I can find on getting WPA2 Enterprise to work and feel like I'm missing something really simple here.  If I go to .../esp8266/tools/sdk/version on my computer, it reads 2.0.0_16_08_09 ... so it appears I did successfully download version 2.0 as required.

I'm a code novice but obviously it can't find a necessary folder/file during the verify step.  Any assistance from anyone is greatly appreciated.

Thanks,
-Chase

adriencapaine, 2017-03-21:
Hi cbrum11,

It looks like the Arduino IDE is looking for the compiler... but at the
wrong path.

Please, ensure that you really have the compiler in *"C:\Program Files
(x86)\Arduino\hardware\esp...\xtensa-lx106-elf-g++".*
In my Windows environement, the compiler is also in
*"C:\Users\<username>\AppData\Local\Arduino15\packages\esp8266com\...".*
I have downloaded and installed the SDK2.0 version in that "packages"
folder.

When you open Arduino and download a random Board with "Tools > Board >
Board Manager", check where that "packages" repository is created. If not
in *"C:\Program Files (x86)\Arduino\hardware\..." *then it was created
somewhere else.

I suspect Arduino to look for the Boards in the "packages" repository
instead of the "Arduino" repository as I can see a few JSON files with the
same classification as in the "Board" Tool in the Arduino IDE.

I have the Arduino 1.6.8 portable version.

I hope this will help ;)

Matthew-Bradley, 2017-03-23:
I've been wondering for a long time now, why is there so much rigamarole involved with getting PEAP working? I could understand if it were merely an incomplete implementation from Espressif, or a wrong one, that there would have to be changes made to the source code that is compiled and uploaded. Perhaps some functions and classes in the wifi C++ files would have to be altered, replaced or finished (or you could just substitute the file with a complete version) but I don't understand why editing a (compiled binary?) file with a hex editor is required, or any of the other tricky steps that seem to be throwing people off. Is it possible that we could get to a point where it's just a matter of replacing a directory full of files, or are we bound to wait until Espressif get around to fixing things?

davisonja, 2017-03-23:
Typically we end up having to resort to editing binaries when we can only
freely acquire the relevant library as a compiled object, and the
distribution licence prohibits distribution of a modified version.
You can do it yourself, but it can't be put into a package because then
it's being distributed. Conceivably a script to automate the process is
acceptable, but licensing isn't my thing :)

cbrum11, 2017-03-24:
@adriencapaine 

Thanks so much for the response!

You are absolutely correct that when I download a random board it creates an instance in....

C:\Users\<username>\AppData\Local\Arduino15\packages...

In fact there are also  esp8266com\esp8266\hardware and tools folders in those directories.  But they are all empty?

I will do my best to reinstall the 2.0 SDK to this packages directory and report back with an answer.  Thanks again for your help... I certainly would have never known to look there.

Matthew-Bradley, 2017-03-24:
So we have the libraries that are compiled and cannot be edited, and they have issues. Will Espressif (or others) eventually fix them? Can we substitute the broken parts (if not the whole library) with parts either written from scratch or equivalents obtained elsewhere? What are the specific issues we're having here with PEAP?

Macci0, 2017-05-08:
Are there any People who have successfully connected to the eduroam network using EAP-PEAPv0 (EAP-MSCHAPv2)???
I was able to install the SDK2.0 with Arduino 1.6.8 but the Code from adriencapaine/ninjabe86 was not working. I was not able to login. I edited the  libwpa2.a file also with my login "..........123456@uzh.ch"

[82 more replies]