esp8266/Arduino #1034: NO  SPIFFS option

asetyde, 2015-11-17:
Can it does a NO SPIFFS option , i lose a 1 mb with actual configuration. I don't use OTA therefore why lost 1/4 of space ?

igrr, 2015-11-18:
I suppose you refer to 4M (1M SPIFFS) configuration.
If you don't use SPIFFS (i.e. if you don't run `SPIFFS.begin()`), then this 1M region of flash is not written, and you can use it as you like.

```
            1M          2M          3M          4M
|-----------|-----------|-----------|-----------|
| sketch->  |           |    <-OTA  | SPIFFS |XX|
                                                `SDK config + EEPROM (5 sectors)
```

asetyde, 2015-11-18:
Ok thank you , yes i speak about this . But if use up 3M , therefore can i  use last 1m if my code is 3,5M ?

igrr, 2015-11-18:
Your code can not exceed 1M. But you can store data in the rest of flash memory.

asetyde, 2015-11-18:
Wait but if with stable (no spiffs) i can use  full 4M ?

At now with staging max 3M ?

igrr, 2015-11-18:
You can not have more than 1M of code in any version (stable, staging, git,
whatever).
This is a limitation of how flash memory is mapped to CPU address space by
the chip itself.

Technically, it is possible to work around this limitation by switching
mapped regions at runtime, but this approach is a bit hard on the user.
This is like multiple code banks on 8051 series MCUs.

asetyde, 2015-11-18:
Ok this config 3M or 1M is better to use . Thank you

closed 2015-11-18 by asetyde