esp8266/Arduino #1035: SimpleAuthentification.ino Headers not  available for GET Request

avarachanvc, 2015-11-17:
I am trying to use the new example code SimpleAuthentification.ino. I See that it has additional functions for parsing headers. But If I simply try to print all the headers for the / (root) page, it only prints User agent, and cookie. I dug into the src, and found parsing.cpp. If I enable debug, it prints all headers on serial but server.header() functions does not seem to get it. Is there a way to enable it so that all headers can be accessed?

avarachanvc, 2015-11-17:
Haha. That one line of code. Thanks!!

avarachanvc, 2015-11-18:
@igrr I went back to what I was trying to do and found no success. I am trying to use the Basic Authentication through the browser header. I updated the code to track "Authentication"
`const char * headerkeys[] = {"User-Agent","Cookie","Authentication"} ;`

Then for a test page, 
`server.on("/pass", handlePassword);`

```
void handlePassword(){
  
  for (uint8_t i=0; i<server.headers(); i++){
    message = " " + server.headerName(i) + ": " + server.header(i) + "\n";
    Serial.print(message);
  }
   if (server.hasHeader("Authentication")){
    Serial.print( server.header("Authentication")) ;
    //I want to check if the base64 encoded usn:pw string matches, but for now, just let in.
    server.send(200, "text/plain", "OK You are in");
  }
  else{
    String header = "HTTP/1.0 401 Authorization Required\r\nWWW-Authenticate: Basic realm=\"Protected\"\r\nContent-Type: text/html\r\n\r\n";
   
    server.sendContent(header);
   // server.send(401, "text/plain", "Need password");
  }
}
```

on /pass, the browser pops the login modal. but when I enter say, admin, password and click login,
it still prints
```
User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/46.0.2490.86 Safari/537.36
 Cookie: 
 Authentication: 
```

I tried calling `server.collectHeaders(headerkeys, headerkeyssize );` inside `handlePassword()` but that did not seem to help. 

Is there something I am missing?

yongkianpeng, 2015-12-19:
Ivan,

When I add the header "Authorization" into the headerkeys array:
const char * headerkeys[] = {"User-Agent","Cookie","Authorization"} ;

I get this error during compilation: 
'class ESP8266WebServer' has no member named 'collectHeaders'

Any idea? Thank you very much.

mkeyno, 2016-02-04:
@igrr  , is there any advance login authentication? any one know someone who did that?

igrr, 2016-03-10:
@kapyaar I tried this and indeed `Authentication` header is empty. However it appears that it is indeed not set by browser. Instead, it sets `Authorization` header.
Here is the serial output with "Debug Level" set to "HTTPServer" in Tools menu:

```
headerName: Host
headerValue: 192.168.222.114
headerName: Connection
headerValue: keep-alive
headerName: Authorization
headerValue: Basic YWFhOmJiYg==
headerName: Accept
headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
headerName: Upgrade-Insecure-Requests
headerValue: 1
headerName: User-Agent
headerValue: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/48.0.2564.116 Safari/537.36
headerName: Accept-Encoding
headerValue: gzip, deflate, sdch
headerName: Accept-Language
headerValue: en-US,en;q=0.8,ru;q=0.6
```

closed 2016-03-10 by igrr

treii28, 2016-08-04:
lulz, I had to giggle when I saw that example sketch. is_authentified? I didn't know the former president was writing example sketches for ESP8266 now. So you run 'is_authentified' when you want to make sure your user is properly authentificated?

luc-github, 2016-08-04:
Yes sorry I used a french based word instead a proper english one when I did it