esp8266/Arduino #1038: Can i use this github without Installing with Boards Manager

Lexander24, 2015-11-18:
Hello, 
I understand that Installing with Boards Manager is make simple to this this code.
But i have problem with company regulation about using this step.

Are available other method for using this code?

Lexander24, 2015-11-18:
I can't download the file online, and searching the offline method

tzapu, 2015-11-18:
you could use the git method, and just take the git folder with you and place it in the proper folder on your company s computer

Lexander24, 2015-11-18:
How to do that?
I download the zip, but got confused about the phyton.

Can you give me instruction to do that

tzapu, 2015-11-18:
Hi, on your local computer, the one you can use on the internet do these steps as per the instructions here:
https://github.com/esp8266/Arduino#using-git-version-

After doing all that, and being able to compile/upload for esp locally, try and move the whole Arduino/hardware/esp8266com/esp8266 to your arduino folder on the restricted computer.

I assume you must have similar configurations on both computers for this to work, same OS, same pythin version and libraries... 

there might be an easier way with whatever the board manager downloads, but i don't know how that would work

Lexander24, 2015-11-18:
I didn't understand the step.
What is mean about this

Clone this repository into hardware/esp8266com/esp8266 directory (or clone it elsewhere and create

Are the clone meaning is download the zip and copy the file to the folder.

And how to use the phyton?

igrr, 2015-11-18:
By clone i mean to run `git clone https://github.com/esp8266/Arduino.git esp8266`.
When you do this, your directory structure should look like:
```
    Arduino
    |
    --- hardware
        |
        +-- arduino
        --- esp8266com
            |
            --- esp8266
                |
                --- bootloaders
                --- cores
                --- doc
                --- libraries
                --- package
                --- tests
                --- tools
                --- variants
                --- platform.txt
                --- programmers.txt
                --- README.md
                --- boards.txt
                --- LICENSE
```
You can also download .zip from github and extract it. But using `git clone` will allow you to update easily later on by doing `git pull`.

When you're done with this step, open terminal/command prompt, go to `hardware/tools` directory and run `python get.py`. This will download all the necessary tools for you. Afterwards just copy the whole directory to the flash drive and move it to the computer which doesn't have Internet access.

closed 2015-11-21 by igrr