esp8266/Arduino #1039: OTA port not visible on OS X

tzapu, 2015-11-18:
hi,
I am trying to get OTA working but can't seem to make arduino ide play ball and see the port.
Here is my setup:
- latest git pull version of the esp package
- arduino ide from http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-4107-BUILD-421-macosx.zip
- basicOTA example sketch
- nodeMCU 0.9 board set to nodeMCU 1.0 to get the OTA option in tools menu

Here is where i got to:
- module connects to wifi and seems to start up ota lib
```
OTA server at: esp8266-a4512a.local:8266
Ready
IP address: 10.0.1.16
```
- can ping it using name 
```
ping esp8266-a4512a.local
PING esp8266-a4512a.local (10.0.1.16): 56 data bytes
64 bytes from 10.0.1.16: icmp_seq=0 ttl=255 time=308.529 
```
- bonjour browser does not see it
- arduino ide does not see it

Not sure what to try next

reaper7, 2015-11-18:
similar problem https://github.com/esp8266/Arduino/issues/970
try add to basicOTA sketch: 
before setup  
    WiFiServer TelnetServer(8266);

and inside setup:
    TelnetServer.begin();

tzapu, 2015-11-18:
Hi, I *thought* that was supposed to be fixed by using the custom built ArduinoIDE.
Anyway, i tried what you said, still did not appear. 
Waited a while, still did not appear.
Run the upload manually `python /Users/alex/Documents/Arduino/hardware/esp8266com/esp8266/tools/espota.py -i esp8266-a4512a.local -p 8266 -f /var/folders/_x/q1phvmbx67s57yzh21bss7g40000gn/T/buildd5cc3e787ad9669d7a177fcef443849b.tmp/BasicOTA.ino.bin` and that WORKED fine
Now the port is also visible in ArduinoIDE.
I removed the the WiFIServer code above, port still there, still uploading.
Restarted IDE, still there, still working.
Rebooted mac, after a short delay, port appeared, still working...

I assume serial console does not work with the ota method?
Cheers

igrr, 2015-11-18:
Serial console doesn't work, it needs more substantial changes to the IDE.

tzapu, 2015-11-18:
thanks
i'll close this now, seems to work for me at least. people can try the steps above if they have similar issues

closed 2015-11-18 by tzapu

me-no-dev, 2015-11-18:
I am experiencing an issue where if I keep the IDE open for some prolonged period of time, it stops finding the boards. If I restart the IDE, they appear as usual. And I'm on a Mac with the captain onboard.

bedenko, 2015-12-03:
I have the same issue. OS Ubuntu 15.10, Arduino 1.6.5 with esp8266_2.0.0 board.
Tried @reaper7  suggestion without success.
Command `avahi-browse -a` does not show me the board (nodemcu 1.0)
Also tried command `ubus call mdns scan` and `ubus call mdns browse` on my Openwrt router, which also does not display the board.

SpaceTeddy, 2016-12-10:
Also issues with Mac OS Ide.
Same sketch on Nodemcu, but IDE under raspberry finds the nodemcu port

CorpusCallosum, 2017-03-09:
I'm having the same issue on Arduino 1.8.1
Network port appears sometimes, sometimes not.

newkit, 2017-09-06:
Same here, restarting the IDE helps for the moment... :-S

tokra, 2020-12-30:
Not working for me :(