esp8266/Arduino #1040: Documentation update in "OTA Update" section

krzychb, 2015-11-18:
Documentation link: https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md 

I am currently preparing an update for ```ESP8266HTTPUpdateServer``` class aka “Updater class” (https://github.com/esp8266/Arduino/pull/1024#commitcomment-14402029) and found a dead link in original table of contents to “Stream Interface”. 

Should “Stream Interface" be deleted from TOC or you like to keep it?

If you like to keep it, could you provide any reference to a library or code to be mentioned or included in doc?

Krzysztof

Links2004, 2015-11-18:
the Stream Interface reveres to:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp#L371

this function is the base for all other update modes like OTA, http Server / client.

me-no-dev, 2015-11-18:
ESP8266HTTPUpdateServer != Updater Class
Updater is in the Core and deals with writing the firmware to the flash, checking it's integrity and telling the bootloader to load the new firmware on the next boot.

krzychb, 2015-11-18:
@me-no-dev - please review my original question https://github.com/esp8266/Arduino/pull/1024#commitcomment-14402413 - I was let alone, assumed something and here we are :smile: 
I have another OTA doc entry almost ready. I should be able to submit it later today so you can check if it is any good.

Krzysztof

closed 2015-11-21 by krzychb