esp8266/Arduino #1041: Problem downloading win32-xtensa-lx106-elf-gb404fb9-2.tar.gz

dahlma, 2015-11-18:
Hello!

I can not install the additional board, the Arduino IDE Board Manager returns error downloading win32-xtensa-lx106-elf-gb404fb9-2.tar.gz and if I try git and get.py it fails on the same file.

This is on Windows 7. 

Any tips how to succeed? 

Please and thank you

closed 2015-11-18 by dahlma

VeggieVampire, 2016-02-05:
delete it from \AppData\Roaming\Arduino15\staging\packages and try re-downloading it from the boards manager

jhendricks123, 2017-04-11:
Sorry for the bump of an old closed thread but just wanted to add a comment here that I had the same issue and the problem was that my companies Sonicwall was blocking the download with the following error:

Gateway Anti-Virus Alert

This request is blocked by the Firewall Gateway Anti-Virus Service. Name: Pirminay.A_19 (Trojan) blocked.

I discovered this by manually trying to download the same files which showed up in the staging/packages directory. The esp8266-2.3.0.zip file downloaded fine, but the win32-xtensa-lx106-elf-gb404fb9-2.tar.gz file was being blocked by the firewall.

I grabbed the download URL's for these files directly from http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json