esp8266/Arduino #1042: ArduinoOTA stops working together with ESP8266WebServer

reaper7, 2015-11-18:
when I try to add WebServer to BasicOTA example then OTA stops working.
I add only (before setup):
```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#include <ESP8266WebServer.h>

ESP8266WebServer server(80); 
``` 
serial console messages when I try upload via OTA:
```
Booting
Ready
IP address: 192.168.4.28
Error[1]: Begin Failed
```
arduino ide message:
```
11:22:26 [ERROR]: No response from device
```

me-no-dev, 2015-11-18:
your sketch becomes too large and there is no space to write the new one.
OTA is really intended for larger Flash chips. 512K is way not enough :)

reaper7, 2015-11-18:
seems to be true :)
without WebServer 245714b(49%)
with      WebServer 247662b(49%)
this is bad news, because OTA could be good solution for some devices like esp-05 (only with rx+tx pins) where programming is little difficult.
I have few esp-05, so...maybe changing flash ic's is last chance for them.
BTW...is any good and tested type of bigger flash memory?

me-no-dev, 2015-11-18:
any 25Q compatible chip should do :) i bought some 8MB Winbond chips that I use (up to 4MB of them).

reaper7, 2015-11-18:
bought W25Q32 :)
tnx!
close!

closed 2015-11-18 by reaper7

reaper7, 2015-12-03:
I only notice that
with latest release size of final sketch is OK:)
without WebServer 222006b(44%)
with WebServer 224094b(44%)