esp8266/Arduino #1045: Insert SPIFFS upload function in your own program?

rudi48, 2015-11-18:
@Ivan , Thank you very much for the excellent programming of the Arduino plugin.

I have tried to incorporate the **SPIFFS upload function** in my own program. I mashed source code from ESP8266WebServer/Examples/**FSBrowser.ino** and **WebUpdate.ino**.

Please see following what I put together:
````c
const char* serverIndex = "<form method='POST' action='/update' enctype='multipart/form-data'><input type='file' name='update'><input type='submit' value='Update'></form>";

void setup() {
  // get filename from client for upload; setup is also handle
  server.on("/upload", HTTP_GET, []() {
    server.sendHeader("Connection", "close");
    server.sendHeader("Access-Control-Allow-Origin", "*");
    server.send(200, "text/html", serverIndex);
  });

  server.onFileUpload([]() {  // called several times from Parsing.cpp while upload
    if (server.uri() != "/update") return;
    // get file
    HTTPUpload& upload = server.upload();
    Serial.println("status: " + (int)upload.status);
    if (upload.status == UPLOAD_FILE_START) {
      fileName = upload.filename;
      Serial.println("Upload Name: " + fileName);
      String path = "/" + fileName;
      UploadFile = SPIFFS.open(path, "w");
      // already existing file will be overwritten!
    } else if (upload.status == UPLOAD_FILE_WRITE) {
      if (UploadFile)
        UploadFile.write(upload.buf, upload.currentSize);
      Serial.println(fileName + " size: " + upload.currentSize);
    } else if (upload.status == UPLOAD_FILE_END) {
      Serial.println("Upload Size: " + upload.totalSize);
      if (UploadFile)
        UploadFile.close();
    }
    yield();
  });

  server.on("/update", HTTP_POST, []() {
    server.sendHeader("Connection", "close");
    server.sendHeader("Access-Control-Allow-Origin", "*");
    server.send(200, "text/plain", (Update.hasError()) ? "FAIL" : "OK");
    // ESP.restart();  // not needed
    DBG_OUTPUT_PORT.println("dir SPIFFS");
    Dir dir = SPIFFS.openDir("/");
    while (dir.next()) {
      fileName = dir.fileName();
      size_t fileSize = dir.fileSize();
      DBG_OUTPUT_PORT.printf("FS File: %s, size: %s\n", fileName.c_str(), formatBytes(fileSize).c_str());
    }
  });

````
**In Principe it works.** What confuses me is the output of the **Serial Monitor**. I have marked with "<---" what I expect:
````c
status: <--- there should be a <number>
Upload Name: favicon.ico
tatus:   <--- should be "status <number>"
favicon.ico size: 318
atus:   <--- should be "status <number>"
svg+xml  <--- I did not program that
        <--- missing: "Upload size: 318"
dir SPIFFS
FS File: /wifi-carAP.js, size: 3.37KB
FS File: /WiFi-car_gzip.png, size: 88.46KB
FS File: /favicon.ico, size: 318B
```` 
**Question:** What I am doing wrong?
Regards, Rudolf
p.s. my IoT project is described here: http://www.rudiswiki.de/wiki9/WiFiCar-NodeMCU

Links2004, 2015-11-19:
you try to add a ```int``` to an ```const char*``` array 
that will not work.
```Serial.println("status: " + (int)upload.status);```
your code moves the pointer of the char by the value of upload.status thats the reason why you get only the half of you text.

to get it work you can do it like this:

```cpp
Serial.print("status: ");
Serial.println((int)upload.status);
```

or (costs more ram)
```cpp
Serial.println(String("status: ") + (int)upload.status);
```

rudi48, 2015-11-19:
@Links2004 
Thank you very much Markus for the help.
Yeah, "C" is not easy, and I am working mostly copy and paste :-)

Your first solution works fine, see the output of the Serial Monitor:
`````c
status: 0
Upload Name: favicon.ico
status: 1
favicon.ico size: 318
status: 2
Upload Size: 318
dir SPIFFS
FS File: /wifi-carAP.js, size: 3.37KB
FS File: /WiFi-car_gzip.png, size: 88.46KB
FS File: /favicon.ico, size: 318B
`````
My other problem was solved the same way with:
`````c
      Serial.print("Upload Size: ");
      Serial.println(upload.totalSize);
````

closed 2015-11-19 by rudi48