esp8266/Arduino #1046: Interrupts and ESP8266WiFi.h

dahlma, 2015-11-18:
Hello!

Why can I not add intterrupts when including ESP8266WiFi.h?

I can compile and run simple code with interrupts. I can compile and run the example wificlient sketch successfully. But when I add an interrupt to a sketch including  ESP8266WiFi.h  (Ex: attachInterrupt(13, onPulse, FALLING);) I get this error when compiling:

WiFiClient:24: error: 'onPulse' was not declared in this scope

Please give me some advice or explanation why the two can not be combined!

igrr, 2015-11-18:
Please attach the complete sketch you are trying to compile. Also do enable
verbose compiler output in Arduino preferences and post full output (not
just the error) as well.

dahlma, 2015-11-19:
WifiClient example with an interrupt added 

/*
    This sketch sends data via HTTP GET requests to data.sparkfun.com service.

    You need to get streamId and privateKey at data.sparkfun.com and paste them
    below. Or just customize this script to talk to other HTTP servers.

*/

#include <ESP8266WiFi.h>

const char* ssid     = "your-ssid";
const char* password = "your-password";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup() {
  Serial.begin(115200);
  delay(10);

  attachInterrupt(13, onPulse, FALLING);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int value = 0;

void loop() {
  delay(5000);
  ++value;

  Serial.print("connecting to ");
  Serial.println(host);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  // We now create a URI for the request
  String url = "/input/";
  url += streamId;
  url += "?private_key=";
  url += privateKey;
  url += "&value=";
  url += value;

[... 30 lines cut ...]
"C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu" "C:\Users\msdgse\AppData\Local\Temp\build663fede588dbec713d8a2ca444eff311.tmp\sketch\WiFiClient.ino.cpp"
"C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src" "C:\Users\msdgse\AppData\Local\Temp\build663fede588dbec713d8a2ca444eff311.tmp\sketch\WiFiClient.ino.cpp"
"C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src" "C:\Users\msdgse\AppData\Local\Temp\build663fede588dbec713d8a2ca444eff311.tmp\sketch\WiFiClient.ino.cpp"
"C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src" "C:\Users\msdgse\AppData\Local\Temp\build663fede588dbec713d8a2ca444eff311.tmp\sketch\WiFiClient.ino.cpp"
"C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src" "C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp"
"C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu" "-IC:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src [... cut ...] h\WiFiClient.ino.cpp.o"
C:\Users\msdgse\AppData\Local\Temp\arduino_663fede588dbec713d8a2ca444eff311\WiFiClient.ino: In function 'void setup()':

WiFiClient:22: error: 'onPulse' was not declared in this scope

   attachInterrupt(13, onPulse, FALLING);

                       ^

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\msdgse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi 
exit status 1
'onPulse' was not declared in this scope

dahlma, 2015-11-19:
adding a void onPulse does not make a difference.

igrr, 2015-11-19:
@dahlma there is no onPulse in your example :)
If you still get an error once you have added onPulse, please update your post accordingly. Thanks.

igrr, 2015-11-19:
This compiles fine:
```c++
#include <ESP8266WiFi.h>

const char* ssid     = "your-ssid";
const char* password = "your-password";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void onPulse();

void setup() {
  Serial.begin(115200);
  delay(10);

  attachInterrupt(13, onPulse, FALLING);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int value = 0;

void loop() {
  delay(5000);
  ++value;

  Serial.print("connecting to ");
  Serial.println(host);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  // We now create a URI for the request
  String url = "/input/";
  url += streamId;
  url += "?private_key=";
  url += privateKey;
  url += "&value=";
  url += value;

  Serial.print("Requesting URL: ");
  Serial.println(url);

  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n\r\n");
  delay(10);

  // Read all the lines of the reply from server and print them to Serial
  while (client.available()) {
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }

  Serial.println();
  Serial.println("closing connection");
}

void onPulse() {
  // do something here
}
```

dahlma, 2015-11-19:
wtf, I must have missed a { or something..

Thanks (feeling like an idiot..)

closed 2015-11-30 by igrr