esp8266/Arduino #1049: Web Server Slow - Quick Fix

hookedupjoe, 2015-11-19:
I noticed the web server was very slow when using lots of writes.  Tracking back the code, I found delay(5000) inside the ClientContext.h.  I changed that to a smaller number and poof - web replies that took 30 seconds reduced to 853ms (granted that had lots of client writes as a test to max out slowness).  I am using this for intranet purposes only, so don't need a long timeout .. not sure what that is used for, etc.   But reporting my findings / changes.

========== Updated the delay(5000) as in ClientContext.h to this ...

            delay(_writeDelay); // max send timeout

========== Added the variable and setter to allow tweaking in ClientContext.h 
		void setWriteDelay(uint16_t theDelay){
			_writeDelay = theDelay;
		}
		
    private:
		uint16_t _writeDelay = 5;

========== 

End of comments - Thanks.

drmpf, 2015-11-19:
Try my non-blocking version of TCP/IP, pfodESP8266WiFi,  and see if that is any better for you.
available from http://www.forward.com.au/pfod/pfodParserLibraries/index.html
with examples.
Same class names just different includes.

hookedupjoe, 2015-11-21:
I tried but get WPS_CB_STATUS_t not found error and can't track it back yet.

closed 2016-02-29 by igrr

Nasgh, 2017-01-19:
hi
where I use this "setWriteDelay()";

J0n4e, 2017-11-02:
@hookedupjoe hookedupjoe - Thanks for the info, works perfectly, i have no more delay on ESP8266 WebServer