esp8266/Arduino #1051: http://arduino.esp8266.com/staging/package_esp8266com_index.json  vanished

aliafshar, 2015-11-19:
And other links on arduino.esp8266.com that are needed for the tools.

igrr, 2015-11-19:
DNS is fixed now. Changes should propagate within two hours.

hookedupjoe, 2015-11-19:
For those with no patience like me ... I was able to (temporarily) use the direct IP to pull the latest .. (104.131.82.128).

Example - pulled down staging from ..
http://104.131.82.128/staging/package_esp8266com_index.json

Moderator: Feel free to remove comment if not a cool thing to do :)

meaepeppe, 2015-11-19:
I still can not download from http://104.131.82.128/staging/package_esp8266com_index.json or the official link. Same 404 error. I just need to wait? Thank for helping.

josuechopite, 2015-11-19:
Hi, brother arduino.esp8266.com is down ...

mlongo95, 2015-11-19:
I'm still not able to access/download both staging or stable packages as of right now

gothambeat, 2015-11-19:
After using the direct IP address, the system complains of other missing files. such as win32-xtensa-lx106-elf-gb404fb9-2.tar.gz  in the same dns path that is missing.

chaeplin, 2015-11-19:
dns has prob still.

dns1.stabletransit.com. and dns2.stabletransit.com. don't have records for arduino.esp8266.com.

;; QUESTION SECTION:
;arduino.esp8266.com.		IN	A

;; AUTHORITY SECTION:
esp8266.com.		300	IN	SOA	dns1.stabletransit.com. ipadmin.stabletransit.com. 1447965043 21600 3600 1814400 300

aliafshar, 2015-11-19:
If you have an old installation somewhere, you can grab the extra tools from:

~/.arduino15/staging/packages/linux64-xtensa-lx106-elf-gb404fb9.tar.gz

And put them in the dist/ directory before running get.py

aliafshar, 2015-11-19:
If you have an installation, you can grab the extra tools from:

~/.arduino15/staging/packages/linux64-xtensa-lx106-elf-gb404fb9.tar.gz etc

claybar, 2015-11-20:
Substituting the IP in the additional boards manager URL allows the Board Manager to update, but does not allow the packages to be installed since the links within the JSON don't have the IP.  Just need to be patient is seems.

Any updates on when it might be fixed?

CurlyWurly-1, 2015-11-20:
using this works for me atm 
http://104.131.82.128/stable/package_esp8266com_index.json

In my case. I had previously installed the ESP8266 stuff a few months back, but when I executed the arduino IDE this morning, the ESP8266 boards had dissappeared.from the list.
I changed the preference link to the above and the boards all came back :)

gothambeat, 2015-11-20:
Is there a way to download the JSON using the direct IP address then edit it to add the same ip address into the file before installing?

mitos14, 2015-11-20:
@CurlyWurly-1 same exact thing happened to me and I am stuck now :(

Links2004, 2015-11-20:
#### work around

add 
```
104.131.82.128 arduino.esp8266.com
```

to your host file at you computer.

```
Windows: %SystemRoot%\system32\drivers\etc\hosts
Linux: /etc/hosts
MAC: /private/etc/hosts
```

mitos14, 2015-11-20:
@Links2004 u are AWESOME. That worked. I have no idea why that never came to mind lol I have confirmed that I can download all packages via arduino IDE and burn a programs that works. I did this on the Adafruit Huzzah

sureshsarat, 2015-11-20:
Did as Links2004 mentioned Downloads fine but on install throws CRC doesn't match error and fails install.

igrr, 2015-11-20:
If you get CRC errors, delete the offending file in
Arduino15/packages/staging.

gothambeat, 2015-11-20:
workaround updating the hosts file worked like a charm.  THANKS!

meaepeppe, 2015-11-20:
Workaround is perfect! Thanks!

Links2004, 2015-11-20:
right now all is looking good:

[image]

asetyde, 2015-11-20:
no stable no staging , i can not download nothing !

Links2004, 2015-11-20:
it need some time before all the different DNS servers get the changes,
the one from google (8.8.8.8) has it.

asetyde, 2015-11-20:
Why this block ? at now ok  thx

abhindas, 2015-11-20:
It worked, but Programmer is not showing esptools as an option, I am using Arduino 1.6.5 on windows 10.

Links2004, 2015-11-20:
@abhindas  check: https://github.com/esp8266/Arduino/issues/739#issuecomment-136739612

sureshsarat, 2015-11-21:
Links2004 Workaround the best. The CRC issue was also resolved by removing /arduino15/.. package files as per IGRR. Thanks to every body

sureshsarat, 2015-11-21:
Sorry for not mentioning earlier. THANKS TO ALL OF YOU

Links2004, 2015-11-21:
the DNS is working fine now again.

### I recommend to remove the workaround from the host file.

closed 2015-11-21 by Links2004

HansZimmer, 2016-02-27:
arduino.esp8266.com is off-line again...

krzychb, 2016-02-27:
arduino.esp8266.com is off-line for me well

HansZimmer, 2016-02-27:
@igrr ?

giolenas, 2016-02-27:
http://arduino.esp8266.com/stable/package_esp8266com_index.json

Is Down!!!!! :(

nagayaoh, 2016-02-27:
Dowin... orz

n-ovesen, 2016-02-27:
yeah, seems to fail here as well

HansZimmer, 2016-02-27:
Is is back

krzychb, 2016-02-27:
Same here (It is back) :smile:

HansZimmer, 2016-02-27:
:)

re3lex, 2016-04-20:
arduino.esp8266.com is down

yorik, 2017-03-11:
arduino.esp8266.com is still down. any news on that?