esp8266/Arduino #1053: Add EEPROM.get() and EEPROM.put()

eliabieri, 2015-11-19:
These two functions allow to write any data type to EEPROM.
I`m not experienced enough. 
Would it be difficult to implement that?

vlast3k, 2015-11-19:
They work, i am using them in my code and it is fine
here you can find some details on how to work with the EEPROM library on the ESP8266
https://github.com/esp8266/Arduino/blob/master/doc/reference.md#eeprom

drmpf, 2015-11-19:
```
// set the EEPROM structure
struct EEPROM_storage {
  uint32_t timeout;
  uint32_t baudRate;
  uint16_t portNo;
  char ssid[pfodESP8266Utils::MAX_SSID_LEN + 1]; // WIFI ssid + null
  char password[pfodESP8266Utils::MAX_PASSWORD_LEN + 1]; // WiFi password,  if empyt use OPEN, else use AUTO (WEP/WPA/WPA2) + null
  char staticIP[pfodESP8266Utils::MAX_STATICIP_LEN + 1]; // staticIP, if empty use DHCP + null
} storage;
const int EEPROM_storageSize = sizeof(EEPROM_storage);
```
Read
```
  uint8_t * byteStorageRead = (uint8_t *)&storage;
  for (size_t i = 0; i < EEPROM_storageSize; i++) {
    byteStorageRead[i] = EEPROM.read(wifiConfigEEPROMStartAddress + i);
  }
```
Write
```
    uint8_t * byteStorage = (uint8_t *)&storage;
    for (size_t i = 0; i < EEPROM_storageSize; i++) {
      EEPROM.write(wifiConfigEEPROMStartAddress + i, byteStorage[i]);
    }
    EEPROM.commit()
```

closed 2015-11-30 by igrr

SimonKlausLudwig, 2017-08-31:
I had to add EEPROM.begin(EEPROM_storageSize) before read and write to get the code from @drmpf working. 

Thank you very much anyway :)