esp8266/Arduino #1057: Can't access URLs at arduino.esp8266.com

kotobuki, 2015-11-20:
Hi,

It seems that URLs at arduino.esp8266.com (e.g. http://arduino.esp8266.com/stable/package_esp8266com_index.json) are not accessible. I have no problem accessing github.com/esp8266/Arduino.

I'm sorry if this is a false alarm or my environment specific issue.

me-no-dev, 2015-11-20:
there are issues with the domain esp8266.com and all underlying subdomains. Should be resolved soon :)

mitos14, 2015-11-20:
I am having the same issue, but I saw others here https://github.com/esp8266/Arduino/issues/1051 going to the IP address directly. When I try to install the esp8266 board from Board Manger, I get this whacky error:
```
CRC doesn't match. File is corrupted.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:170)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:164)
	... 1 more
 CRC doesn't match. File is corrupted.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:170)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:164)
	... 1 more
 CRC doesn't match. File is corrupted.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:170)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$24(ContributionManagerUI.java:164)
	... 1 more
```

Any help would be greatly appreciated

EskimoRuler, 2015-11-20:
I edited my host file in windows for arduino.esp8266.com and pointed it to 104.131.82.128 and I was able to download in board manager

ry60003333, 2015-11-20:
Any ETA for when this will be resolved?

igrr, 2015-11-20:
No ETA for now.

wich0, 2015-11-20:
sorry for my bad english, but i just download the file package_esp8266 com_index.json and replace all http://arduino.esp8266.com/ to http://104.131.82.128/ and save it on a local server as ngnix, I did that and it worked fine for me

Testato, 2015-11-20:
it's down from two day ?

@igrr why not leave the board manager json and all the related core file on GitHub ?

mitos14, 2015-11-20:
@wich0 I tried to do what u did (I am on OSX Yosemite). I installed nginx and made the changes to the json file. I verified that I can fetch the json file from the browser. Now I see this error in my arduino IDE:
```
http://localhost:8080/package_esp8266com_index.json file signature verification failed. File ignored.
```
Did you have to do anything to get around this error? I wish I can figure out how to produce that signature file myself

thank you

Juppit, 2015-11-20:
@mitos14 My you have a look in build_boards_manager_package.sh it's in 
hardware/esp8266com/esp8266/packages

On 20.11.2015 at 16:26 wrote mitos14:

dvdavide, 2015-11-20:
until the DNS issues get sorted out, you might try manually adding the line:
`104.131.82.128  arduino.esp8266.com`
to your hosts file. (it's under C:\Windows\System32\drivers\etc in windows)
Worked for me.

Links2004, 2015-11-20:
duplicate of: #1051

check here workaround here:
https://github.com/esp8266/Arduino/issues/1051#issuecomment-158434266

will close this to track the problem in one issues.

closed 2015-11-20 by Links2004