esp8266/Arduino #1059: After update to last staging eclipse give -    error: arduino.ar: No such file or directory

far5893, 2015-11-20:
Starting archiver
"/home/user/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/media/user/74a50ba9-aa9c-4031-8341-b77f685e75f6/cvs/progetti/esp8266/esp8266-WebServices/Release/core.a"    ./arduino/spiffs_cache.c.o ./arduino/spiffs_check.c.o ./arduino/spiffs_gc.c.o ./arduino/spiffs_hydrogen.c.o ./arduino/spiffs_nucleus.c.o  ./arduino/Esp.cpp.o ./arduino/FS.cpp.o ./arduino/HardwareSerial.cpp.o ./arduino/IPAddress.cpp.o ./arduino/MD5Builder.cpp.o ./arduino/Print.cpp.o ./arduino/Stream.cpp.o ./arduino/Tone.cpp.o ./arduino/Updater.cpp.o ./arduino/WMath.cpp.o ./arduino/WString.cpp.o ./arduino/abi.cpp.o ./arduino/cont.S.o ./arduino/cont_util.c.o ./arduino/core_esp8266_eboot_command.c.o ./arduino/core_esp8266_flash_utils.c.o ./arduino/core_esp8266_i2s.c.o ./arduino/core_esp8266_main.cpp.o ./arduino/core_esp8266_noniso.c.o ./arduino/core_esp8266_phy.c.o ./arduino/core_esp8266_postmortem.c.o ./arduino/core_esp8266_si2c.c.o ./arduino/core_esp8266_timer.c.o ./arduino/core_esp8266_wiring.c.o ./arduino/core_esp8266_wiring_analog.c.o ./arduino/core_esp8266_wiring_digital.c.o ./arduino/core_esp8266_wiring_pulse.c.o ./arduino/core_esp8266_wiring_pwm.c.o ./arduino/core_esp8266_wiring_shift.c.o ./arduino/debug.cpp.o ./arduino/libc_replacements.c.o ./arduino/pgmspace.cpp.o ./arduino/spiffs_api.cpp.o ./arduino/spiffs_hal.cpp.o ./arduino/time.c.o 
Finished building: arduino.ar
 
Starting combiner
"/home/user/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-L/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc1/tools/sdk/lib" "-L/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.0.0-rc1/tools/sdk/ld" "-Teagle.flash.512k0.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "/media/user/74a50ba9-aa9c-4031-8341-b77f685e75f6/cvs/progetti/esp8266/esp8266-WebServices/Release/esp8266_WebServices.elf" -Wl,--start-group    ./.ino.cpp.o ./BME280.cpp.o ./SFE_BMP180.cpp.o ./StringCircularBuffer.cpp.o ./ajax.js.cpp.o ./ambient.cpp.o ./io.html.cpp.o ./oled.cpp.o ./standard.css.cpp.o  ./Libraries/Wire/Wire.cpp.o  ./Libraries/Ticker/Ticker.cpp.o  ./Libraries/TSL2561/TSL2561.cpp.o  ./Libraries/SI7021/SI7021.cpp.o  ./Libraries/Rtc/RtcDS1307.cpp.o ./Libraries/Rtc/RtcDS3231.cpp.o ./Libraries/Rtc/RtcDateTime.cpp.o ./Libraries/Rtc/RtcUtility.cpp.o  ./Libraries/I2C/EEPROM24.cpp.o ./Libraries/I2C/I2CMaster.cpp.o ./Libraries/I2C/SoftI2C.cpp.o  ./Libraries/GOFi2cOLED/GOFi2cOLED.cpp.o  ./Libraries/ESP8266mDNS/ESP8266mDNS.cpp.o  ./Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o ./Libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp.o ./Libraries/ESP8266WiFi/src/WiFiClient.cpp.o ./Libraries/ESP8266WiFi/src/WiFiClientSecure.cpp.o ./Libraries/ESP8266WiFi/src/WiFiServer.cpp.o ./Libraries/ESP8266WiFi/src/WiFiUdp.cpp.o  ./Libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.o ./Libraries/ESP8266WebServer/src/Parsing.cpp.o  ./Libraries/EEPROM/EEPROM.cpp.o   arduino.ar   "/media/user/74a50ba9-aa9c-4031-8341-b77f685e75f6/cvs/progetti/esp8266/esp8266-WebServices/Release/core.a" -lm -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -lwps -lcrypto -laxtls -Wl,--end-group  "-L/media/user/74a50ba9-aa9c-4031-8341-b77f685e75f6/cvs/progetti/esp8266/esp8266-WebServices/Release"
xtensa-lx106-elf-gcc: error: arduino.ar: No such file or directory
make: *** [esp8266_WebServices.elf] Error 1

igrr, 2015-11-20:
Should be fixed by 7137ec5.

far5893, 2015-11-20:
Hand modify  platform.txt fix the bug. 
Can Close

closed 2015-11-20 by Links2004