esp8266/Arduino #1060: Unable to get Board details under board manager

sureshsarat, 2015-11-20:
Arduino 1.6.5r5 installed.
I used all possible values http://arduino.esp8266.com/stable/package_esp8266com_index.json &http://arduino.esp8266.com/staging/package_esp8266com_index.json.
I am getting signature verification failed. File ignored.ESP8266 not showing up in Arduino

FYI
NODEMCU DEVKIT 1 used

sureshsarat, 2015-11-20:
Closing as this post has the solution http://arduino.esp8266.com/staging/package_esp8266com_index.json vanished #1051

Using Ip ADDRESS

sureshsarat, 2015-11-20:
Ok Though I am able to get listing in Boards Manager installation fails with CRC error.

Links2004, 2015-11-20:
duplicate of: #1051

check workaround here:
https://github.com/esp8266/Arduino/issues/1051#issuecomment-158434266

for CRC error check:
https://github.com/esp8266/Arduino/issues/1051#issuecomment-158451098

will close this to track the problem in one issues.

closed 2015-11-20 by Links2004