esp8266/Arduino #1061: Captive Portal SSID behaviour on apple devices

avarachanvc, 2015-11-20:
In captive portal
```
    String ssidString = "Setup-" + String(system_get_chip_id());
    const char *ssid = ssidString.c_str();
    WiFi.softAP(ssid);
```
Does not popup the captiveportal window on apple devices. 
`WiFi.softAP("Setup-12345');`

Works without any issues. Has anyone else observed this behaviour?

me-no-dev, 2015-11-20:
try 
```cpp
String ssidString = "Setup-" + String(system_get_chip_id());
WiFi.softAP((const char *)ssidString.c_str());
```

avarachanvc, 2015-11-20:
Thanks for the suggestion. Tried it, it works on iphone now, but not on macbook. Windows has no issues with any method though.

devyte, 2017-10-19:
I can't reproduce this in latest git. Closing.

closed 2017-10-19 by devyte