esp8266/Arduino #1062: Hardware SPI

Shibonja, 2015-11-20:
Hi guys,

First of all Thanks for the good work, you made development much easier with esp integration with Arduino IDE. 
I have noticed that the ESP-12E has hardware SPI pins, so my question is: "has the Arduino SPI library support for these pins, or is the library using bit-banging with digital pins."

Thanks in advance, Haris.

igrr, 2015-11-20:
SPI library uses hardware HSPI (a.k.a. SPI1) peripheral (pins 12-15).
Extra pins on ESP-12E are connected to SPI0 peripheral, which is not supported by SPI0 library at the moment.

Shibonja, 2015-11-20:
Hi @igrr ,
Thanks for quick answer.
Are you planning to add support for them in future releases?

igrr, 2015-11-20:
Actually I did implement this in a hacky manner about a month ago, need to revisit these changes and clean up the code.

Shibonja, 2015-11-20:
Hi @igrr ,
I ordered a few ESP8226-12E, so if u need any help, maybe I can be at your service. I am not a C/C++ guru but I can find my way around.

Links2004, 2015-11-20:
is SPI0 not shared with the flash?
so all the SPI code need to be in ram,
may gives problems with other interrupts too?

igrr, 2015-11-20:
Actually this is done as follows: you are still using SPI1 peripheral (i.e. registers are the same), but its pins are mapped to the same pins as SPI0. There is a hardware arbiter which can demultiplex two SPI peripherals to the same physical set of pins, controlling two CS lines automatically. SPI0 has higher priority, so flash requests are still serviced, and you can use cached code. Access times via SPI1 become a bit unpredictable in this mode, but you can have an SPI peripheral and keep pins 12-15 free at the same time.

reaper7, 2015-11-22:
@igrr - very interesting...is there any chance for tests this solution?

steffenmauch, 2015-12-14:
@igrr Is there any possiblity to test your solution?

KaloNK, 2015-12-23:
+1 for this feature.

Will it be possible to switch SPI1 between hardware and pins 12-15 at runtime?

I was looking for something similar to https://github.com/espressif/esp8266_mp3_decoder implementation for SPI SRAM (or [FRAM](https://github.com/adafruit/Adafruit_FRAM_SPI) ) chip access.
The idea is to use the same pins in SQI mode or better yet in SDI mode with the CS0 attached to the HOLD, so flash access will not interfere with a RAM stream if interrupted and still have the pins 12-15 as dedicated SPI for other devices ... as if there are 3 SPI:
SPI0 - flash
SPI1 - user
SPI2 - RAM in SDI mode, paralel to the flash, but with user defined CS

KaloNK, 2016-01-13:
> Actually I did implement this in a hacky manner about a month ago, need to revisit these changes and clean up the code.

@igrr shouldn't it be enough to call hspi_dev_sel() ( esp_iot_sdk
\examples\driver_lib\driver\spi_overlap.c ) inside beginTransaction()?
It can actually replace the <code>while(SPI1CMD & SPIBUSY) {}</code> call as it uses WAIT_HSPI_IDLE() macro which does the same.
So when we use HSPI we call hspi_dev_sel(0) and for RAM we call hspi_dev_sel(2) which will use GPIO0 (1 for GPIO1 is an option, but as it is TX0 can't be used). Settings should be extended with the mode (DIO or QIO) in addition to the spi dev used.

If such approach is acceptable I can try to modify the SPI driver in a backward compatible way and post a pull request with the changes.

KaloNK, 2016-01-26:
I have made a patch to implement this based on the SDK example code https://github.com/KaloNK/Arduino/commit/48ef8c434678ebd41cbdcc750ab0859489efbcd9 but it is not tested yet, because I am still waiting for the SRAM chip.
I am about to add support for (S/F/M)RAM attached parallel to the flash in (S/D/Q)IO modes too, but i guess it should be separate patch

whhsn, 2016-07-19:
Check this:

https://hackaday.io/project/8678-rpi-wifi/log/32008-on-sdio-and-the-mysteries-of-the-universe

KaloNK, 2016-07-19:
I have some code [here](https://github.com/KaloNK/Arduino/tree/ExtendSPI) and [here](https://github.com/KaloNK/Arduino/tree/SPI_RAM), but couldn't make it work with SRAM ( 23LC1024 ) attached to the same bus as flash (it does not read anything from SPI0 for some reason). I do not have a logic analyzer, so it is difficult to debug what is happening.
I will come back to this some time later, but if someone can review the code and see a mistake there causing it not to work, will help a lot

igrr, 2016-07-19:
@KaloNK, thank you for your work.
FYI, there is a $250 [bounty](https://www.bountysource.com/issues/28455602-support-for-spi0-in-spi-library) on this issue.

mkeyno, 2016-07-20:
@igrr  can you elaborate, regards to this topic, what feature will be added to the ESP8266? if new HW SPI will be available for user , it is only break out on ESP-12E and not break out for most current available module in market

boks21, 2017-01-03:
Hi,
 im' new using esp8266 with spi( i used it with uart)

Now i need use esp12E with (H)SPI and im doing first steps and have some doubts:

Its factory ready to use HSPI? or i need to reprogram?
Some special pin conections to use HSPI?

Someone knows registers adress to write/read data to Wi-Fi?

Thanks a lot,

Loureiro

pipi61, 2017-01-03:
http://www.esp8266.com/viewtopic.php?f=32&t=11396&p=54834&hilit=spi+gpio+12+13+14#p54834

igrr, 2017-05-08:
Implemented in #3189.

@riataman please go to bountysource to claim your bounty.

closed 2017-05-08 by igrr

WillyKirsch, 2017-07-16:
Hi, I'm trying to use the SPI library to read an SD card, with the previous version (without the PSI.pins() method) and worked fine, now with the SPI.Pins() method, I'm having this error when compile:

...esp8266\2.3.0\libraries\SPI\SPI.cpp:132:52: error: 'SPIPCS2DIS' was not declared in this scope

                 SPI1P |= SPIPCS1DIS | SPIPCS0DIS | SPIPCS2DIS;

in any line where that names are used.

can anybody help me?

thanks!!

usonix, 2018-02-05:
Hi, I'm new here and also not very experienced in programming, I read what is described above, but it was not very clear to me. What I need is to use the SD card on the SPI0 pins, leaving the SPI1 pins free for other applications, if not ask too much, could someone publish an example code how to do this?

riataman, 2018-02-05:
I never got around getting a SD card working in SPI0 (I did got a SPI memory workking), I think the sd card library would need changes to make it work. 

I didn't had the proper tools at the time to try and debug this and right now I lack to time :(

javadhabibi1500, 2019-09-20:
Except power-up, when the esp needs to use flash memory through SPi0? In other words, when my overlaped SPI is busy?