esp8266/Arduino #1063: Eclipse data folder to spiffs  upload.

far5893, 2015-11-20:
Under eclipse compiled with (512k/64k) option  there is no way  to build and flash  data directory.

igrr, 2015-11-20:
Do you mean you want SPIFFS upload function implemented in Eclipse?
Eclipse isn't officially supported, but I think @Links2004 is using it for development. Perhaps he can give some advice.

Links2004, 2015-11-20:
you can create your own make targets for example mkspiffs.

1. create a file ```makefile.init``` in the project root

```makefile
vecho := @echo
Q := @

ARDUINO_BASE      = D:/Coding/avr/Programme/arduino-nightly
ESP8266_BASE      = $(ARDUINO_BASE)/hardware/esp8266com/esp8266
ESP8266_TOOLS     = $(ESP8266_BASE)/tools

mkspiffs:
	$(vecho) Staring mkspiffs...
	$(ESP8266_TOOLS)/mkspiffs/mkspiffs.exe -c "../data" mkspiffs.bin

```

2.
[image]

3.
[image]

@igrr can you check the mkspiff command, i never used it before ;) is the block, size and page needed?
 
@far5893 may your ESP8266_BASE path is in your user arduino dir (when you have installed via json)

igrr, 2015-11-20:
You also need to pass block size, page size, and image size to mkspiffs.
These are defined for each flash size variant in boards.txt, i.e. for 512k(64k SPIFFS):
```
generic.menu.FlashSize.512K64.build.spiffs_start=0x6B000
generic.menu.FlashSize.512K64.build.spiffs_end=0x7B000
generic.menu.FlashSize.512K64.build.spiffs_blocksize=4096
```

These values must be given to mkspiffs, otherwise created image will not be compatible with SPIFFS configuration in firmware.

mkspiffs command line arguments are described here: https://github.com/igrr/mkspiffs#usage

devyte, 2017-10-19:
Closing per #3655 .

closed 2017-10-19 by devyte