esp8266/Arduino #1064: Implement a way to post functions to the main loop

igrr, 2015-11-20:
This should provide an easy way to trigger blocking code from interrupts and timers without having to pass events through flags.

baruch, 2016-06-19:
Why not make this a library? as it is it will be a burden to any system since it runs in the main outside of the loop. If someone wants this functionality is it so hard to add a call to a library inside the loop to run this?

igrr, 2016-06-20:
Currently the "burden" is one extra function call and one `if (ptr !=0)` check, plus a few global variables. I wouldn't exactly call that a burden, compared to all other stuff which happens in `main`.

baruch, 2016-06-20:
It also pulls all the code and space needed for the list of function pointers. It's not a big burden but ram is not so plentiful that it can be completely ignored. I'm not going to press this in any way, it's a design philosophy more than anything else.

closed 2016-06-23 by igrr