esp8266/Arduino #1065: Wifi strange error

asetyde, 2015-11-20:
After changed from stable to new staging 2.0 i 've seen strange wifi error , at 17.30~ my nodemcu every 3 day stopped to work , after i rebooted (power off to power on ) go normally , no error see !

Strange i must to come back to stable .. please an adivice , less plus future more stabilizing this good work .. 

in stable i need reser manual after every flash with new no , good work

marvinroger, 2015-11-21:
Mine has been running fine for a week, with the latest git rev as of 9/14. Maybe you're experiencing a memory leak?

asetyde, 2015-11-21:
I don't know , i see esp connected but no way to response . I want to try monitoring serial but 24h is impossible .. Now I try with stable , if work we know or it's my code .

Another things I see many type of wifi monitoring and start , no a standard can do some mistake. 

Now I try with first test and we see. 

Thx for reply , stay tuned ;)

asetyde, 2015-11-22:
No version seems problem .. at now i think my stability function can be ?

void wifi() { //- wifi control & stability v.1.1
  
  if (crazycounter > 40) { diagnostic(8); }

  if (WiFi.status() == WL_CONNECT_FAILED) {
      diagnostic(9);
    }

    if (WiFi.status() == WL_CONNECTION_LOST) {
      diagnostic(4);
      delay(2000);
      restart();
    }

    if ((WiFi.status() == WL_DISCONNECTED) { diagnostic(4); delay(2000); restart(); }

  }
}

asetyde, 2015-11-22:
Test :   
with stable - > after some hours wifi sys freeze (with wifi() function FAIL  , without -waiting to test- )
with 1.6.5 i hope !  without wifi function -testing now- , with wifi() -waiting to test-

with 2.0.0 -> after 2/3 min freeze with wifi and without FAIL

asetyde, 2015-11-23:
After upgrade my esp seems broken , and now i would ask @igrr  if it is possible ?  , at now code is blocked random in ten minutes , 4 day ago work perfectly

asetyde, 2015-11-24:
At now , on stable i resolve strange error , depends from sensor , i don't know if is port or library .
On staging confirm not work , start and block . Hope be resolved.

asetyde, 2015-11-24:
https://github.com/esp8266/Arduino/issues/940

asetyde, 2015-11-24:
I probably found a bug , with my mod dht library 1.2.3 can block device , now investigating , i restore dht standard 1.1.1 and try . my device has 2 type of sensor dht and analogo photo-resistor , i think it's impossibile that photo-resistor can block device , i suppose this .

i ll give news as soon as possible

asetyde, 2015-11-26:
ok problem come from dh library . please see #940

closed 2015-11-26 by asetyde