esp8266/Arduino #1066: Arduino 1.6.6 now requires function prototypes in sketches (preprocessor changed?)

ladyada, 2015-11-21:
or at least, all my code didnt work without em :)

e.g.
`setup() {
// stuff
}

loop() {
// do things
 myfunction();
}

void myfunction() {
//code
}`

no longer works without sticking `void myfunction();`
at the top (before loop())
is it a big deal? not really, but its ... unusual since neither avr compilation requires nor was this required before.
Its mentioned here https://www.arduino.cc/en/Main/FAQ
"Your sketch undergoes minor changes (e.g. automatic generation of function prototypes) and then is passed directly to a C/C++ compiler (avr-g++)."
perhaps that step is not done anymore?
anyways, ill look into it too

krzychb, 2015-11-21:
I have noticed the same with [Arduino-PR-4107-BUILD-421](https://github.com/esp8266/Arduino/pull/984#issuecomment-155905800) but assumed this was a special build.
Krzysztof

ladyada, 2015-11-21:
ok yeah its a thing thats now done with arduino-builder 
https://github.com/arduino/arduino-builder
(still doing some investigation but its gonna slow down because i dont speak fluent go)

ladyada, 2015-11-21:
ok so its actually not even getting to the prototype gen state. what happens is it hits the 
`#include "ESP8266WiFi.h"`
and fails/crashes - never continuing on to the prototype generation stage

igrr, 2015-11-21:
Traced to `#include <memory>` in WiFiClient.h.
Workaround is possible.

neuhaus, 2015-12-21:
I'm affected by this issue, too.
However, the title is wrong - this is not an issue with Arduino 1.6.6, it's an issue with ESP8266 Version 2.0. I was having this problem with Arduino 1.6.5.

sdamaggio, 2016-02-01:
I confirm this behaviour. I'm using Arduino IDE 1.6.7 with esp8266 boards package version 2.0.0.

igrr, 2016-02-01:
This is fixed now in Arduino nightly builds, see https://github.com/arduino/arduino-builder/issues/68.

closed 2016-02-27 by igrr