esp8266/Arduino #1068: OTA update result 115200 baud output on uart0.

EUA, 2015-11-22:
Flashed chip with OTA (while debug enabled), using 74880 baud.
After successful OTA session, somehow, device starts with 115200 baud that seen garbage as output.

````
OTA server at: esp8266-gatekeeper-devel.local:8266
sleep enable,type: 2
Ready
Update Start: ip:192.168.1.2, port:51400, size:228640, md5:1de0582b748efc1d769dcc0341f16d24
sleep disable
Update Success
Rebooting...
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/26053553
del if0
usl
sul 0 0
"@*rjrA(!�S�u�К��XKA�~��Q���UK��I*E,�-1��Ъ¡��A�!��n�*E�HhI�A,T[M�vA,T[�����p*�scandone
f 0, scandone
state: 0 -> 2 (b0)
reconnect
state: 2 -> 0 (0)
f -180, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 
````

Garbaged section at 115200 baud:
````
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld
````

Don't understand why it switches to 115200 baud itself after OTA program.

me-no-dev, 2015-11-22:
What you see is the output of eboot. Looking at the eboot source, it does not start the UART (I think it was starting it in some older revision), but it writes to it. I will go wild and say that the UART keeps those bytes in it's internal FIFO and as soon as the firmware starts and loads the UART, those bytes come to you. Arduino ESP starts at 115200bps (see core_esp8266_main.cpp) and that happens a bit before it gets to the setup() and give you the option to alter the bitrate.
If you hit the reset button, you should see the output of eboot at 74880 ecause the ESP's own bootloader (in ROM) has started first and booted the UART at 74880. OTA is using software reset which does not trigger the onboard bootloader :)

me-no-dev, 2015-11-22:
as I'm thinking more about it, when I OTA new firmware i see the output of eboot streaming, so data is not kept in FIFO, but something has set the baudrate to 115200 before eboot is loaded and maybe even before the software reset triggered (since you see the boot mode message).
@igrr any chance the ESP jumped to "user_init()" and then interrupted to actually reset when ESP.restart() is called? I do not see the baudrate set to 115200 anywhere else

igrr, 2015-11-22:
I have never seen such behavior, and I find it difficult to understand how
the code could jump to user_init. Perhaps there's something in user's code
which triggers this.

me-no-dev, 2015-11-22:
@EUA which version of the core are you running? stable/staging/git?
can you please share the OTA part of your sketch?
@igrr what will happen if ESP.reset() is called right after ESP.restart() or the other way around?

EUA, 2015-11-22:
@me-no-dev  I use latest git.
It doesn't need to upload my sketch.
It's possible to replicate this behavior with ArduinoOTA->BasicOTA example.
Just change baud line as 74880
````
void setup() {
  Serial.begin(74880);
  Serial.println("Booting");
````
Results:
````
Progress: 99%
Progress: 100%
End
�@�rjrA(!�S�u�P���XKA�~��Q���UK��I*E,�-1��Ъ¡��A����n�*E"�hI�A,T[M�vA,T[�`��,p*�Booting
Ready
IP address: 192.168.1.117
````

igrr, 2015-11-22:
You cannot call anything after ESP.restart, because ESP.restart switches to OS context and Arduino context no longer runs. So unless you call reset from a timer or gpio interrupt, this should not happen.

EUA, 2015-11-23:
@me-no-dev , Changing core_esp8266_main.cpp file baud rate from 115200 to 74880 doesn't fix the issue.

devyte, 2018-02-28:
Closing due to lack of feedback and age.

closed 2018-02-28 by devyte