esp8266/Arduino #1070: memory leek by using WiFiClient

Links2004, 2015-11-22:
test code:
```cpp
void loop() {
    USE_SERIAL.printf("RAM: %d\n", ESP.getFreeHeap());
    // wait for WiFi connection
    if((WiFi.status() == WL_CONNECTED)) {
        WiFiClient * _tcp = new WiFiClient();
        _tcp->connect("192.168.1.12", 80);
        _tcp->stop();
        _tcp->~WiFiClient(); //does not help but shut abort any connection and free all
    }
    delay(100);
}
```

return:
```
RAM: 41312
RAM: 41312
RAM: 41048
RAM: 40816
RAM: 40584
RAM: 40352
RAM: 40104
RAM: 39872
RAM: 39640
RAM: 39408
RAM: 39176
RAM: 38944
RAM: 38712
RAM: 38480
RAM: 38248
RAM: 38016
RAM: 37784
```

igrr, 2015-11-22:
https://github.com/esp8266/Arduino/issues/230#issuecomment-101371864

igrr, 2015-11-22:
You are doing active close. Send Connection: close header and wait for server to do active close.

Links2004, 2015-11-22:
can confirm this but the active close takes very long (60sec).
but i still miss 376Byte.

start: 41296
avter close timeout: 40920 

````cpp
void loop() {
    static int i = 10;
    static uint32_t last_ram = 0;
    uint32_t ram = ESP.getFreeHeap();
    USE_SERIAL.printf("RAM: %d (%d)\n", ram, (ram - last_ram));
    last_ram = ram;
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED) && ram > 2000 && --i > 0) {

        WiFiClient * _tcp = new WiFiClient();
        _tcp->connect("192.168.1.12", 80);
        _tcp->stop();
        _tcp->~WiFiClient(); //does not help but shut abort any connection and free all
        delay(100);
    } else {
        delay(1000);
    }
}
```

````
[SETUP] WAIT 2...
[SETUP] WAIT 1...
RAM: 41296
RAM: 41296 (41296)
RAM: 41032 (-264)
RAM: 40800 (-232)
RAM: 40568 (-232)
RAM: 40336 (-232)
RAM: 40088 (-248)
RAM: 39856 (-232)
RAM: 39624 (-232)
RAM: 39392 (-232)
RAM: 39160 (-232)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 39160 (0)
RAM: 40696 (1536) <-- active close end?
RAM: 40920 (224)  <-- active close end?
RAM: 40920 (0)
RAM: 40920 (0)
RAM: 40920 (0)
RAM: 40920 (0)
RAM: 40920 (0)
RAM: 40920 (0)
RAM: 40920 (0)
```

Links2004, 2015-11-22:
ok the server was not sending an ```ACK``` message to the ```FIN``` message, seams to be a anti flood protection if I add a ```_tcp->println("GET / HTTP/1.1");``` I instand become the ```ACK``` for the close

but I still miss ~360 Byte.
eave after several minutes they not come back.

```
[SETUP] WAIT 2...
[SETUP] WAIT 1...
RAM: 41280
RAM: 41280 (41280)
RAM: 41208 (-72)
RAM: 41200 (-8)
RAM: 41128 (-72)
RAM: 41120 (-8)
RAM: 41048 (-72)
RAM: 41008 (-40)
RAM: 41000 (-8)
RAM: 40928 (-72)
RAM: 40920 (-8)
RAM: 40920 (0)
RAM: 40920 (0)
....
```

Links2004, 2015-11-25:
Note:
# Never call the deconstructor directly! use delete!

good:
```cpp
void loop() {
    USE_SERIAL.printf("RAM: %d\n", ESP.getFreeHeap());
    // wait for WiFi connection
    if((WiFi.status() == WL_CONNECTED)) {
        WiFiClient * _tcp = new WiFiClient();
        _tcp->connect("192.168.1.12", 80);
        _tcp->stop();
        delete _tcp;
    }
    delay(100);
}
```

closed 2015-11-25 by Links2004

laercionit, 2020-03-15:
I have this problem at the moment, using core 2.6.3, is this still in master or has it already been covered in this core?