esp8266/Arduino #1071: WDT Reset PubSub MQTT upon disconnection, or reconnection.

Cosford, 2015-11-22:
Hi, I'm working on developing an application using an ESP8266 and the PubSubClient library, however in testing the robustness of my code, I've found that closing the broker that the client is connected to (in this case, I'm hosting a Mosquitto broker on a local PC), results in the ESP8266 immediately ceasing responding; soon followed by a watchdog timer reset. (This is running the esp8266 example provided in the pubsubclient library).
Some experimentation has found the issue is also apparent if the WiFi connection is disconnected, then reconnected; resulting in a lock up.

I've posted a similar issue with the pubsubclient library here (https://github.com/knolleary/pubsubclient/issues/95) but I suspect the issue is to do with the WiFiClient object itself?

Any ideas? Thanks in advance.

marvinroger, 2015-11-24:
I don't have such an issue with my network-tolerant code. Here is how I do:

```c++
void wifiConnect() {
  WiFi.hostname("esp8266");
  WiFi.mode(WIFI_STA);
  WiFi.begin("SSID", "Password");
}

void mqttConnect() {
  mqtt->setServer("192.168.0.10", 1883);
  mqtt->setCallback(mqttCallback);

  if (mqtt->connect("esp8266")) {
    mqttSetup();
  }
}

void mqttSetup() {
  mqtt->subscribe("input_topic", 1);
}

void mqttCallback(char* topic, byte* payload, unsigned int length) {
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    unsigned long now = millis();
    if (now - last_wifi_reconnect_attempt > 20000UL || last_wifi_reconnect_attempt == 0) {
      Serial.println("Attempting to connect to WiFi");
      last_wifi_reconnect_attempt = now;
      wifiConnect();
    }
    return;
  }

  if (!mqtt->connected()) {
    unsigned long now = millis();
    if (now - last_mqtt_reconnect_attempt > 5000UL || last_mqtt_reconnect_attempt == 0) {
      Serial.println("Attempting to connect to MQTT");
      last_mqtt_reconnect_attempt = now;
      mqttConnect();
    }
    return;
  }

  // Arrived here, we are connected to both Wi-Fi and MQTT

}
```

It's been up for a week and I disconnected my router and my broker a lot of time to check the robustness of the code, no issues so far.

Cosford, 2015-11-25:
Thanks for your reply. Your code here looks remarkably similar to my own with regard to checking WiFi.status() and mqtt->connected().

May I ask for a complete sketch and what versions of libraries you're using as I can't understand why my code is throwing an issue when yours is not?

marvinroger, 2015-11-25:
Here is the sketch: http://pastebin.com/pe40dbmd
I am using Arduino 1.6.6 with ESP8266 hardware git rev https://github.com/esp8266/Arduino/commit/590eefa210dfb5e519859d85cd51c6b3e12ab9dd (see https://github.com/esp8266/Arduino#using-git-version- for installation) with PubSubClient 2.3 (https://github.com/knolleary/pubsubclient/releases/tag/v2.3). 

My hardware is NodeMCU v1.0.

marvinroger, 2015-11-25:
Note: in the sketch, the LED is ON when trying to connect to Wi-Fi/MQTT, it is off when everything is connected.

Cosford, 2015-11-25:
Excellent! Thanks for that! I got the most recent staging version from the boards manager and, at least at first glance, seems to have solved my issues! Thanks very much!

closed 2015-11-25 by Cosford