esp8266/Arduino #1072: pulseIn does not return 0 on timeout (it returns 1 or timeout value depending on pin state)

darkskiez, 2015-11-22:
In core_esp8266_wiring_pulse.c, it seems to return duration=timeout rather than duration=0 on a timeout, or, because it resets the start time after a timeout, it can also return duration=1.

darkskiez, 2015-11-23:
Whilst you're in that file, you could remove the pinMode setting, I may have set PULLUP etc.

framb00s, 2016-01-18:
I have the same problem
const int dustPin= D2;
pinMode(dustPin,INPUT_PULLUP);
int i=0;
void loop() {
  duration = pulseIn(dustPin, LOW);
  if(duration!=0){                                                    // number positive readings
    i++;
  }
  lowpulseoccupancy = lowpulseoccupancy+duration; 
.....
give with nothing attached after 30 seconds 30 reads (i==30)
and a total pulse-low_time between 30 and 35

devyte, 2017-10-19:
@igrr the referenced PR was merged quite some time ago. Can this be closed, or is there something pending?

devyte, 2017-10-19:
@me-no-dev I just realized this is assigned to you, maybe you're the right person to ask?

me-no-dev, 2017-10-19:
Yeah I have no clue how and when I assigned this to myself. I guess it should be closed, unless the current implementation also does not work

devyte, 2017-10-19:
Alright, closing. If there is still a problem, let's handle it in a new issue.

closed 2017-10-19 by devyte