esp8266/Arduino #1073: Exception(0) when switching from station mode to acces point mode.

Duality4Y, 2015-11-23:
hi there!
my code start the esp in station mode, and there is a button on my board and when i press it,
i want the esp to go to access point mode.

but when i try and switch to access point mode it crashes and prints:
```mode is now AP_MODE

0.0.0.0
state: 5 -> 0 (0)
rm 0
pm close 7 0 0
/7892663
del if0
usl
sul 0 0
mode : softAP(1a:fe:34:f3:f7:c6)
add if1
dhcp server start:(ip:1
Exception (0):
epc1=0x40208668 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdc0:  00380031 ffffffff 3fffc250 4000050c  
3ffffdd0:  400043df 00000030 00000016 ffffffff  
3ffffde0:  400044ab 3fffc718 3ffffed0 08000000  
3ffffdf0:  60000200 08000000 00000003 00000000  
3ffffe00:  0000ffff 00000001 04000002 003fd000  
3ffffe10:  3fff3a78 000003fd 3ffef204 00000030  
3ffffe20:  40224f4b 3ffef034 3ffee701 3ffef270  
3ffffe30:  60000600 3ffef4c4 00000000 3ffffdc0  
3ffffe40:  0000112c 402250fc 3ffee71d 3ffffee8  
3ffffe50:  3fff37e0 60000600 3ffffee8 3fff37e0  
3ffffe60:  4022512c 3ffee6ff 3ffffee8 ffffff80  
3ffffe70:  40225fc0 40227312 3ffee6f4 ffffff80  
3ffffe80:  3ffffec8 00000010 402274b5 3ffedc30  
3ffffe90:  3fff37e0 3ffe9d60 60000600 3fff3360  
3ffffea0:  3ffedc30 3ffedc30 3fff37e0 3ffe9d60  
3ffffeb0:  3ffef588 4010136b f334fe1a fffeffff  
3ffffec0:  00000001 00000000 40223a2a 3fff37e0  
3ffffed0:  00000002 4000410f 3ffef034 3fff37e0  
3ffffee0:  3fffc718 40004a3c 000003fd 3fff3a78  
3ffffef0:  3fffc718 40101648 00000378 3ffee6f4  
3fffff00:  000003fd 4021372f 00000378 000003ff  
3fffff10:  00001000 402137c5 3ffef204 000003ff  
3fffff20:  000003fd 60000600 003fd000 3ffef204  
3fffff30:  ffffff00 55aa55aa 00000160 0000001c  
3fffff40:  0000001c 00000076 000000fe 000003ff  
3fffff50:  4021366c 3ffe9c30 00000000 3ffe9d60  
3fffff60:  00000001 40212731 00000002 ffffffff  
3fffff70:  402124b9 3ffe9c24 3ffe9bd0 023d6e19  
3fffff80:  402124fe 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe9be8 3ffea520 00000005 40207419  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

```

so when i try and see what is happening at that point i do:
```
$ xtensa-lx106-elf-objdump -x EspLight-firmware.elf -d | grep -i 40208668
40208668 g     F .irom0.text	0000000a _ZN14HardwareSerial11isRxEnabledEv
40208668 <_ZN14HardwareSerial11isRxEnabledEv>:
40208668:	5228      	l32i.n	a2, a2, 20
```

but I do not understand what this meens and how I might solve this problem with this dump.

igrr, 2015-11-23:
This may happen if UART interrupt comes while ESP is switching between modes.
Will fix.

igrr, 2015-11-23:
Actually looking at the code it appears that `wifi_set_opmode` is already protected from UART interrupts.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L94-L99
So this shouldn't be happening.
Could you upload your sketch and .elf file somewhere so that i can take a closer look at the stack?

Duality4Y, 2015-11-23:
The sketch can be found here:
https://github.com/EspLight/EspLight-firmware
and the binary + .elf here:
https://github.com/EspLight/EspLight-firmware/tree/master/bin

igrr, 2015-12-06:
Was that dump taken with the same .elf which is available in your repository? I tried to load it but addresses do not seem to match.

Duality4Y, 2015-12-08:
had to recompile to create the elf, maybe something happend ?

Links2004, 2015-12-31:
please retest with the latest git.
WiFi management is has been reworked #1323

Duality4Y, 2016-01-01:
yes I saw a bigg pull request!
I will try and test it.

igrr, 2016-02-27:
@Duality4Y were you able to reproduce the issue?

Duality4Y, 2016-02-27:
the issue seems to have gone now.

closed 2016-02-27 by Duality4Y