esp8266/Arduino #1074: Uploads and runs ok, but does not run after power cycle

funvill, 2015-11-23:
Hello 

I have a NodeMCU devkit 0.9 that I am programming with the Arduino IDE using the ESP8266 library. 

I can upload the blink sketch to the NodeMCU, and after its done uploading the blink sketch works without an issue. But when I power cycle the board the the NodeMCU doesn't run blink sketch. It looks like it loses its program... 

I can reprogram the board over and over again and it works until I power cycle the board. 

Any thoughts on what might be causing this ?

funvill, 2015-11-23:
After reading this post 
https://github.com/esp8266/Arduino/issues/1017#issuecomment-156689684

I connected ESP8266 GPIO15 (nodeMCU=D1) to GND. Now I can program the NodeMCU and the program continues to run after a power cycle. 

But when I open a serial connection via the USB serial interface on the NodeMCU the sketch stops processing and I don't' get any output on the serial interface. I use the serial interface for debugging.

asetyde, 2015-11-23:
> #1017 (comment)

Also I , my program work with stable , with new staging blocked after 1/4min , serial dead.

Waiting reply

funvill, 2015-11-24:
To add a little more to this case. 

With ESP8266 GPIO05 (nodeMCU=D1) connected to GND I can program the NodeMCU. After its done uploading but before I restart the NodeMCU I can see the output/debugging info on the USB serial port. 

If I cycle power but do not connect to the USB serial port. The device operates as expected. 
If I cycle power and connect via the USB serial port. The device loses its sketch and I have to reprogram it to get it to work again. 

If i program without the ESP8266 GPIO05 (nodeMCU=D1) connected to GND. Then when I cycle power, regardless if I connect via USB Serial. the device will lose its sketch and will not operate as expected. 

I am using the Blink Sketch from the ESP8266 example code folder while doing my tests.

asetyde, 2015-11-24:
I Think you board is corrupted. 
My nodemcu 1.0 work perfectly no lose of data

Links2004, 2015-11-24:
If i program without the ESP8266 GPIO05 (nodeMCU=D1) connected to GND. Then when I cycle power, regardless if I connect via USB Serial. the device will lose its sketch and will not operate as expected.

are you sure that you not in bootloader mode?
check out here:
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-messages-and-modes

bennthomsen, 2016-05-23:
I suspect the problem occurs because you have connected an led, short or other component from one of the GPIO pins that is also connected to the External Flash memory on the ESP12-E module that the NodeMCU uses. If any of the GPIO pins that are also connected to the flash chip are pulled low then the program will not load from the flash memory when the power cycles. If you want to connect additional components to these pins then add these to the high side. For example to add an LED connect it from Vdd to the GPIO pin (ideally via a current limiting resistor). GPIO0 (nodeMCU=D3) is connected to the external flash.

stubaby49, 2016-08-19:
I just thought I would add my experience of this problem. I have 2 Geekcreit ESP12E DEVKIT V2s, both bought fro Banggood but at different times so they may be different batches. One would run the downloaded program, without the USB cable being connected, after power cycling the other wouldn't.  Both worked perfectly with the USB cable connected. I tried adding a 2200uf electrolytic and 0,1uF disc capacitor to the external 5V  to no avail. I could sometimes get the faulty one to run by applying 5V to the device (it would not run) and then quickly disconnecting and re-applying the 5V. Maybe something on the board was charging up, and if power was re-applied while still charged, it ran. Sometimes this worked consistently, sometime not at all ?????
Then after trying all this stuff I managed to connect 5V to one of the GND pins. This released the blue smoke from one of the components near the blue (blinky) LED. Now the board will not work with BOTH the USB and external 5V supply connected (it pulls the external supply down) BUT I can download program with only the USB connected and run downloaded programs with only the external 5V connected. It runs  EVERY time I re-apply the external power..
I know this isn't a solution but my experience may add to the knowledge pool.

ajinkyaw, 2016-10-27:
hi guys.
i am very very new to esp8266 and these things.
i have learned my self to program arduino and esp8266 device.
currently i am using esp8266-12e . 
whenever i try to upload the code to esp8266 via TTL , it successfully gets uploaded and runs immediately as expected . but when i turn off the power supply and turn it on again . esp8266 doesnt work . there i need to upload the program again. 
suggestions needed.
thanks in advance.
my pin configs are 

vcc - vcc(3,3v)
gnd - gnd
txd - rxd (ttl)
rxd - txd (ttl)
gpio 0 - gnd
gpio15 - gnd
en - vcc

grissk, 2016-11-05:
Hi ajinkyaw,

Once programmed, you need to connect GPIO0 to 3.3v.  Your uploaded code should then run on powerup.

As per https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-messages-and-modes

ajinkyaw, 2016-11-06:
@grissk  : Your solution worked. thanks a ton.

bennthomsen, 2016-11-06:
Note GPIO0 should be pulled up to 3.3V via a 10k resistor.

As per https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-messages-and-modes.

It should not be directly connected to 3.3V. If directly connected to 3.3V and the program tries to pull the pin low you will effectively short the power supply.

zizyx, 2016-12-20:
@bennthomsen "Note GPIO0 should be pulled up to 3.3V via a 10k resistor."

This made my day, ESP kept resetting after I turned of the power or pushed the reset button!! Many thanks

AlbertPhan, 2017-01-31:
I also have NodeMCU devkit 0.9 (off of Ebay with the CH340G TTL-USB chip) and I'm still having this problem despite having the correct pull up resistor on GPIO0. Flashing works fine and it runs the sketch after it's programmed but not when power cycled via USB. For some reason, if I power cycle without USB data (by plugging in USB partially in so data lines aren't connected) the board starts up properly and runs the sketch. If I plug in the USB, it doesn't run the sketch. Something to do with the usb data perhaps? I tried another board and it has the same problem. The 26 MHZ oscillation is present on GPIO0 when the sketch isnt running. Really scratching my head on this one.

rodmcm, 2017-01-31:
Hi
I also have this problem with 2 out of 3 ESP8266 recently purchased. I am using a fidti programmer set to 3.3V and have all the necessary pull up resistors in place. After repowering the units stay in 'Upload' mode. I have reflashed one of them with the latest from Expressif, get all the usual diagnostics when connected but no joy. even after a reflash.. Is it something to do with the arduino loader not setting something so that powerup enables flash memory location?

MoegerleStephan, 2017-05-26:
Hi there,
I'm sadly having the same issue.. are there any solutions till now?
For delaying the start-up process I've also added a RC network with a 10k resistor and a 0.47uF capacitor at the enable pin.
The GPIO15 pin is connected trough a 2.7k resistor to ground. In addition at the GPIO15 is also a LED connected trough a 100 Ohm resistor to ground.
That's a really annoying issue... the often it starts correctly but sometimes not (if not -> the blue flashing LED of the ESP12-f is shining constantly and sometimes  it's completely Off) 

> One would run the downloaded program, without the USB cable being connected, after power cycling the other wouldn't. Both worked perfectly with the USB cable connected. I tried adding a 2200uf [...]
Then after trying all this stuff I managed to connect 5V to one of the GND pins. This released the blue smoke from one of the components near the blue (blinky) LED. Now the board will not work with BOTH the USB and external 5V supply connected

BeePartOfIt, 2017-05-28:
Hi all,
same problem like stubaby49.
Did someone find a solution?
Thanks!

BeePartOfIt, 2017-05-31:
Found something that helped me.
Hopefully someone out there too:

Ok. I figured out my problem. I was using BUILTIN_LED as a flasher to 
indicate when it woke up. In the arduino esp8266 library, that pin is 
connected to io16.... The wake up pin. So when that flashed, it reset 
the board continually.

tukaianirban, 2017-06-12:
Hi,

I am a newbie to ESP8266, and would really appreciate some help here.
I am using a NodeMCU v3 board to try to blink a LED.

Problem : Whenever the board is power cycled, it seems to lose the program on its flash.

My connections are:

(NodeMCU pins)        ( Equipment)
D1 (GPIO5) -------------- (+) LED (-) -------- GND
D3 (GPIO0) -------------- 10K Resistor ------ 3.3V      (used as pull up resistor)
G --------------------------------------------- Power supply GND
Vin ------------------------------------------- 3.3V ( optionally )

Procedure I am following and what I observe :
1. Plug out the Vin connection on NodeMCU.
2. Plugged in serial USB cable to the NodeMCU.
3. From Arduino IDE, uploaded the following sketch.
4. LED starts blinking. Serial output shows up in Arduino IDE.
5. Plug out the serial cable from NodeMCU.
6. Plug in the power supply connection to Vin pin.
7. The LED does not blink any more. Nothing on the Arduino serial monitor either.

With the pull-up 10K resistor connected, D3 always stays at about 2.7V. I dont disconnect / reconnect this at any time. The Vin pin when connected in step#7 receives about 3.3V also.

What could possibly be the problem ? 

Sketch used:
==========================================
int ledpin = 5;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  pinMode(ledpin,OUTPUT);
  
}

void loop() {
  // put your main code here, to run repeatedly:

  digitalWrite(ledpin, HIGH);
  Serial.println("LED glowing");
  delay(1000);
  digitalWrite(ledpin, LOW);
  Serial.println("LED off");
  delay(1000);
}
==========================================

Any help is highly appreciated here.

Thanks,
Anirban

tukaianirban, 2017-06-12:
Hello again,

I figured out the way to contain the program on the flash over a restart.
Found a simple table online:
GPIO15(D8)			GPIO0(D3)			GPIO2(D4)			Mode
0V					0V					3.3V				Uart Bootloader
0V					3.3V				        3.3V				Boot sketch
3.3V				        x					  x				SDIO mode (not used for Arduino)

The 3.3V to the corresponding pins must be going via a 10K pull-up resistor.

Sorry, if I am spamming.

JanuaryThomas, 2017-07-01:
After uploading the program, Connect GPIO0 to Vcc via 10K resistor

sbmmkl, 2017-09-04:
@JanuaryThomas removing all connection and than connecting GPIO 0 to VCC with 10k Resistor ?
my connection is 
GPIO 0 to GND
GPIO 15 to GND
VCC to 3.3v
EN(CH PD) to 3.3v
GROUND to GND
Tx to Rx
Rx to Tx

callmeharry3, 2017-09-09:
I am having the same issues, and would love to solve them.
My board is a wemos d1 r2 I purchased from banggood. I am having some problems checking if it's authentic because my board disconnects everytime I upload a sketch. So that means no serial debugging.

I am trying to make a simple button sketch at first, so if I press a button a LED lights up.
This works if I flash the sketch to the wemos, but after a power cycle I lose the sketch.

My wiring is as follows (I have a normal D1 instead of a mini, but the idea is the same):

[image]

My code is this:

const int D5buttonPin = 14;     // the number of the pushbutton pin
const int D0ledPin =  16;      // the number of the LED pin

// variables will change:
int buttonState = 0;         // variable for reading the pushbutton status

void setup() {
  // initialize the LED pin as an output:
  pinMode(D0ledPin, OUTPUT);
  // initialize the pushbutton pin as an input:
  pinMode(D5buttonPin, INPUT);
}

void loop() {
  // read the state of the pushbutton value:
  buttonState = digitalRead(D5buttonPin);

  // check if the pushbutton is pressed.
  // if it is, the buttonState is HIGH:
  if (buttonState == HIGH) {
    // turn LED on:
    digitalWrite(D0ledPin, HIGH);
  } else {
    // turn LED off:
    digitalWrite(D0ledPin, LOW);
  }
}

I am using the arduino environment (1.8.1) to program my board.

So I am getting the same problems as described above, a power cycle means I lose my sketch. In addition to that, a flash disconnects the board from my COM port.

My questions are:
- which mode do I have to put my board in to prevent sketch loss after a power cycle;
- how do I put my board into that mode? By wiring? By initializing pins in the code? If by wiring, how do I wire them? With resistors, or without?

petpot, 2017-09-11:
Problem solved (nearly). Not the smartest way, but helps to find the final way. 
Power only from extra power supply, power to ESP from USB (not from PC).
Simple and fanny solution ?

[image]

petpot, 2017-09-11:
ewe

devyte, 2017-10-19:
Closing per #3655 .

closed 2017-10-19 by devyte

kovachwt, 2018-01-11:
Anybody still having issues with normal boot, please read this first:
http://www.instructables.com/id/ESP8266-Using-GPIO0-GPIO2-as-inputs/

TLDR: don't use GPIO0 (D3), GPIO2 (D4) and GPIO15 (D8) unless you understand how...

Conorol98, 2018-12-07:
I have found if you connect a button to the between the reset button and GND and reset the serial will work perfectly.

skjolddesign, 2018-12-09:
Had same issue on ESP8266 TTGO  0.91" OLED.
Fixed boot problem with 220 ohm pull-up resistor on GPIO4 "SUA" pin.
1K resistor worked also, but 1.2K did not.
[image]

lrmoreno007, 2018-12-09:
@skjolddesign There are two versions of TTGO and many erroneous pinout graphics, surely you have this one:
[image]

GPIO4 have nothing to do with boot. SUA really is SDA but have a serigraphy error.

skjolddesign, 2019-01-12:
For reference, Recently I got hands on White TTGO. It has this pinout graphics.
Not any boot problem on this one yet.
[image]