esp8266/Arduino #1078: digitalRead() reading intermittently HIGH

Pfannex, 2015-11-24:
Hi,

I use the following code within the main loop,
PIN 5 is used with a pulldown resistor:

```
  #define movement 5
  pinMode(movement, INPUT);
  int movement_state = LOW;

//==============================================================================
//  Main Loop
//==============================================================================

void loop() {

//Bewegungsmelder
  buttonState = LOW;
  buttonState = digitalRead(movement);
  if (buttonState != movement_state) {
    if (buttonState == HIGH){
      Serial.println("movement ++++");
      publish_value(publish_sensor_topic[2], "on");
    }
    else {
      Serial.println("movement ----");
      publish_value(publish_sensor_topic[2], "off");
    }     
    delay(100);                                
  }
  movement_state = buttonState;
}
```

My problem is, that sometimes the digitalRead() will detect a HIGH signal.
Without a HIGH signal on the PIN.

I use a ESP8266-12e with this Chip the wrong HIGH will detected every 10 minutes,
so it looks like a software problem, but I did not used any 10 minutes delay or timer.

[image]

With a ESP8266-12 the problem came sporadic and more often.

Did anyone heared something like that?

asetyde, 2015-11-24:
about use of gpio 5 , also i’ve found some strange issue , using dht on 5 board after some hours go crashed

Pfannex, 2015-11-24:
what about other pins?

drmpf, 2015-11-25:
If you are really connected to a button try adding a debounce to it
see  http://www.forward.com.au/pfod/ArduinoProgramming/DebouncedSwitch/DebouncedSwitch.html

igrr, 2015-11-25:
@Pfannex if you have two ESPs, running the same program, and one shows this behaviour while other doesn't, then this is probably a hardware issue.

Pfannex, 2015-12-04:
Hi,
I found my problem.....
The used sensor got a 3V3 output, but needs 5V operating Voltage.
I don't know why this provokes exactly a HIGH signal every 10 minutes, but now it seems to work.

Thank for your patience.....
Pf@nne

closed 2015-12-04 by Pfannex