esp8266/Arduino #1086: ISR is executing only once.

pbecchi, 2015-11-27:
To test interrupts on GPIO lines I am running this very simple sketch:

void setup() {

  Serial.begin(9600);
  pinMode(13, INPUT);
  digitalWrite(13,HIGH);
  attachInterrupt(13, count0,CHANGE);

Serial.println("Start...");
}

volatile int incr=0,intern=0,flash=0;
int counto=0;
///////////////////////////////////////////////////////////

void count0()
{
  flash++;
 }

void loop() {
 if(flash>counto) {
     counto=flash;
     Serial.print('.');
 /*
    digitalWrite(FPIN, HIGH);   // set the LED on
    delay(50);              // wait for a second
    digitalWrite(FPIN, LOW); 
    delay(50);*/
}
}

The interrupt routine it is executed only first time the interrupt is triggered!

Links2004, 2015-11-27:
```digitalWrite``` to in input pin is not working,
```digitalWrite``` is only for output control.

I suppose you want enable the pullup then use: 
```cpp
pinMode(13, INPUT_PULLUP);
```

pbecchi, 2015-11-27:
Thanks!
But this correction doesn't solve the problem!

pbecchi, 2015-11-27:
I was wrong!
 I have made more testing:
now ISR run many times!!
Thanks!

closed 2015-11-27 by pbecchi