esp8266/Arduino #1090: SPIFFS and EEPROM works together ?

hallard, 2015-11-27:
Hi guys,
I've noticed a strange things and before going to deep dive I wanted to have guru's vision

I've got a sketch I use for long time now that save configuration C structure to EEPROM. This config structure is saved to EEP on change (with crc check) and loaded at startup (again checking CRC). These piece of code already worked well (it comes from my old Arduino code) and I ported on ESP with minor change such as EEP.commit() and was worked well on this target.
Since now I'm using SPIFFS to serve WEB pages, this code does not seems to works anymore my save function always return error after reading back again with CRC check.

I say again, I didn't touched any line of code of this EEP save/load/check routine. So my question is more :+1
Is using EEP and SPIFFS on same time is supported (of course I think so) ? but my 2nd one is
Anyone has tried and succeded on this ?

FYI my config size is 1024 bytes and my code looks like :

```cpp
// Config for emoncms
// 128 Bytes
typedef struct 
{
  char  host[CFG_EMON_HOST_SIZE+1]; 		// FQDN 
  char  apikey[CFG_EMON_APIKEY_SIZE+1]; // Secret
  char  url[CFG_EMON_URL_SIZE+1];  			// Post URL
  uint8_t port;    									    // Protocol port (HTTP/HTTPS)
  uint8_t node;     									  // optional node
  uint32_t freq;                        // refresh rate
  uint8_t filler[23];									  // in case adding data in config avoiding loosing current conf by bad crc*/
} _emoncms;

// Config saved into eeprom
// 1024 bytes total including CRC
typedef struct 
{
  char  ssid[CFG_SSID_SIZE+1]; 		 // SSID     
  char  psk[CFG_PSK_SIZE+1]; 		   // Pre shared key
  char  host[CFG_HOSTNAME_SIZE+1]; // Hostname 
  char  ap_psk[CFG_PSK_SIZE+1];    // Access Point Pre shared key
  char  ota_auth[CFG_PSK_SIZE+1];  // OTA Authentication password
  uint32_t config;           		   // Bit field register 
  uint16_t ota_port;         		   // OTA port 
  _emoncms emoncms;							   // Emoncms configuration
  uint8_t  filler[119];      		   // in case adding data in config avoiding loosing current conf by bad crc
  uint8_t  filler1[512];           // Another filler in case we need more
  uint16_t crc;
} _Config;

// Configuration structure for whole program
_Config config;

uint16_t crc16Update(uint16_t crc, uint8_t a)
{
  int i;
  crc ^= a;
  for (i = 0; i < 8; ++i)  {
    if (crc & 1)
      crc = (crc >> 1) ^ 0xA001;
    else
      crc = (crc >> 1);
  }
  return crc;
}

bool readConfig (bool clear_on_error) 
{
	uint16_t crc = ~0;
	uint8_t * pconfig = (uint8_t *) &config ;
	uint8_t data ;

	// For whole size of config structure
	for (uint16_t i = 0; i < sizeof(_Config); ++i) {
		// read data
		data = EEPROM.read(i);
		
		// save into struct
		*pconfig++ = data ;
		
		// calc CRC
		crc = crc16Update(crc, data);
	}
	
	// CRC Error ?
	if (crc != 0) {
		// Clear config if wanted
    if (clear_on_error)
		  memset(&config, 0, sizeof( _Config ));
		return false;
	}
	
	return true ;
}

bool saveConfig (void) 
{
  uint8_t * pconfig ;
  bool ret_code;

  // Init pointer 
  pconfig = (uint8_t *) &config ;
	
	// Init CRC
  config.crc = ~0;

	// For whole size of config structure, pre-calculate CRC
  for (uint16_t i = 0; i < sizeof (_Config) - 2; ++i)
    config.crc = crc16Update(config.crc, *pconfig++);

	// Re init pointer 
  pconfig = (uint8_t *) &config ;

  // For whole size of config structure, write to EEP
  for (uint16_t i = 0; i < sizeof(_Config); ++i) 
    EEPROM.write(i, *pconfig++);

  // Physically save
  EEPROM.commit();
  
  // Read Again to see if saved ok, but do 
  // not clear if error this avoid clearing
  // default config and breaks OTA
  ret_code = readConfig(false);
  
  Debug(F("Write config"));
  
  if (ret_code)
    Debugln(F("OK!"));
  else
    Debugln(F("Error!"));
  
  // return result
  return (ret_code);
}

void setup(void) {
// Our configuration is stored into EEPROM
  EEPROM.begin(sizeof(_Config));

  DebugF("Config size=");
  Debugln(sizeof(_Config));
  Debugflush();

  // Check File system init 

  if (!SPIFFS.begin())
  {
    // Serious problem
    DebuglnF("SPIFFS Mount failed");
  } else {
   
    DebuglnF("SPIFFS Mount succesfull");

    Dir dir = SPIFFS.openDir("/");
    while (dir.next()) {    
      String fileName = dir.fileName();
      size_t fileSize = dir.fileSize();
      Debugf("FS File: %s, size: %d\n", fileName.c_str(), fileSize);
    }
    DebuglnF("");
  }
  
  // Read Configuration from EEP
  if (readConfig()) {
      DebuglnF("Good CRC, not set!");
  } else {
    // Reset Configuration
    ResetConfig();

    // save back
    saveConfig();

    // Indicate the error in global flags
    config.config |= CFG_BAD_CRC;

    DebuglnF("Reset to default");
  }

bla bla bla
```

igrr, 2015-11-27:
It should work, in the sense that EEPROM and SPIFFS occupy different sectors.
Which build variant (flash size/spiffs size) did you use when you saw this issue? Perhaps some sizes/addresses are not set correctly in the linker script.

hallard, 2015-11-27:
@igrr 
thanks, for this really fast response !
Oh I'm on ESP12 generic 4M/1M SPIFFS and 1024 bytes EEProm
I'm working always on latest repo (I pulled it yesterday by night)

hallard, 2015-11-27:
Ok, I got it, I can't remember if I changed something to struc saved into eeeprom but I've calculated is to be the same size, but I forgot to set aligment on 1 bytes boundary was causing data not to be on the correct place (CRC), I added pragma directive, all is ok now

```
#pragma pack(push)  // push current alignment to stack
#pragma pack(1)     // set alignment to 1 byte boundary

// Config for emoncms
// 128 Bytes
typedef struct 
{
blah 
blah
}
#pragma pack(pop)
```

closed 2015-11-27 by hallard