esp8266/Arduino #1092: OTA dump with ESP-07

SensorsIot, 2015-11-27:
I use the DNS-SD_Arduino_OTA sketch to experiment with OTA. It works fine with nodemcu 1.0 and with ESP-12E. However, with ESP--07s it dumps right after the first 22 points in the IDE. The Message usually is:

Arduino OTA Test
Sketch size: 346656
                   Free size: 2797568
 IP address: 192.168.1.47
Update Start: ip:192.168.1.39, port:48266, size:346672

Exception (9):
epc1=0x401015df epc2=0x00000000 epc3=0x00000000 excvaddr=0x0bf0fe7b depc=0x00000000

ctx: sys
sp: 3ffffce0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe80:  40230eb8 3fff1174 00002200 3fff3a40
3ffffe90:  4020c049 3fff3710 3fff3710 40232644
3ffffea0:  00000000 00000000 3fff1174 40232799
3ffffeb0:  00004089 3fff271c 00000006 00000001
3ffffec0:  40000f58 00000000 40218e43 00000001
3ffffed0:  ffffffff 00000000 3ffec781 00000008
3ffffee0:  40218e8e 3ffefe70 3fff1474 3fff08f0
3ffffef0:  3fff15f4 00000000 40103f23 3ffefe70
3fffff00:  00000000 00000000 00000020 40103e13
3fffff10:  402183d4 3ffefe70 3fff0d70 4020ac79
3fffff20:  4023970e 3fff19bc 3fff3190 3fff1b20
3fffff30:  00000000 3fff3b00 3fff1110 40233b53
3fffff40:  40217e89 3ffec4a0 3ffebe10 00000001
3fffff50:  40217f6c 3ffec4a4 3ffebe10 0109623e
3fffff60:  3fff3070 3fff08f0 3fff090b 60000600
3fffff70:  4021215d 3ffebe10 3ffec1ac 4021215d
3fffff80:  402121a2 3fffdab0 00000000 3fffdcb0
3fffff90:  3ffebe30 3fffdc20 3ffebd6c 40201795
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42

I also saw a few times exception (28), but did not record the dump..

It seems to be ESP-07 specific. I use Generic  ESP8266 board for upload and 4M/1M. The serial upload works ok on these modules.

krzychb, 2015-11-27:
@SensorsIot - I am wondering what is the flash size of your ESP-07. Basing on [this post](http://www.esp8266.com/viewtopic.php?f=32&t=6801) it looks like it is 1M, not 4M. I would try serial upload with 1M/64K selected and then do the same with OTA.
Krzysztof

SensorsIot, 2015-11-27:
@krzychb - You are absolutely right! I was fooled by the "Free size: 2797568" of the OTA sketch. Obviously this number is not calculated from the real flash size. The ESP.getFlashChipSize() shows 524288. This beginners mistake took me now several hours. You saved my weekend! Thanks

closed 2015-11-28 by igrr

mkeyno, 2016-04-10:
hi @SensorsIot  , I have problem to programming ESP-07 ,sometimes is working sometimes is not and not even programmed, can you tell me which IDE setting you were used for and if you have any schematic I really appreciated if you could share it

SensorsIot, 2016-05-03:
@mkeyno There are many tutorials (also on youtube) around on how to program ESP modules. There is not a short answer.

HobbytronicsPK, 2017-12-25:
Correct, OTA works while uploading sketch using 1M(64k SPIFFS).