esp8266/Arduino #1093: Mesh Mode non functional

ibaranov-cp, 2015-11-27:
Hello all,

Using latest staging version, RC2 from board manager.
I have two identical ESP8266-12E modules.
I load them both with the example Mesh code.
I add in a Serial.println("."); into the loop to ensure they are both working.

Both modules wait endlessly.

Does this feature work at all?

igrr, 2015-11-28:
Softap is broken in 2.0.0-rc2 (see #1054), i think this might be the reason for mesh library to have issues.
That said, the author of mesh library doesn't seem to be around any more, so it isn't really maintained any more.

ibaranov-cp, 2015-11-30:
Issue #1054 doesn't seem to be the cause here, I have the fixed version.
Will try and figure out root cause here.

battuashwik, 2016-07-12:
Any luck with the mesh library ?

Makuna, 2016-09-11:
The Esp8266WifiMesh library is not really a good implementation.  Due to being completely open, how synchronous it is written, and that it requires string messages (memory fragmentation from string class and binary data transmission requires conversion to text), its not something to base any serious project on. 

It maybe great to play with though.

devyte, 2017-10-19:
@ibaranov-cp is this issue still valid with latest git?

ibaranov-cp, 2017-10-19:
Unknown, sorry I don't have bandwidth to test this in the near future.

devyte, 2017-10-19:
Ok, closing due to age. If there is still a problem, please open a new issue.

closed 2017-10-19 by devyte