esp8266/Arduino #1094: AP Mode does not allow connection for most devices

ibaranov-cp, 2015-11-27:
Hello all,

Using latest staging version, RC2 from board manager.
I have two identical ESP8266-12E modules. Powered with 600mA 3.3V regulator off of USB V+
I load them both with the example WifiAccessPoint code.
I add in a Serial.println("."); into the loop to ensure they are both working.

When attempting to connect to the modules, phone has to re-connect ~3-5 times to get a connection. And then loading the example webpage at 192.168.4.1 takes ~30 seconds.

Three different models of laptops refuse to connect at all.

comino, 2015-11-27:
Similar observations:
- Im using the latest git version
- ESP-WROOM module
- Hardware is proven to work with SDK example

Tested: 
Lennovo E550- Intel WiFi adapter - no connection at all
Lennovo X Series - after several attemps succesful
Galaxy Tab 2 - successful 
Motorola Moto G - successful
Toshiba Sattelite - Realtek Wifi adapter - successful on first try

me-no-dev, 2015-11-28:
does it make a difference if the AP is Open or Password protected?

comino, 2015-11-28:
It asks me for a password before not connecting

igrr, 2015-11-30:
Since SDK 1.3, RF calibration is stored in flash. On a few occasions i had to wipe flash completely because the board failed to connect to APs.
It would be nice to check if this issue is resolved by wiping RF calibration data. If it is, perhaps it would be worth changing the default RF calibration mode.

ibaranov-cp, 2015-11-30:
I will certainly try, any quick link to wiping the data?

Can't seem to capture any packets on Wireshark during connection, and open vs. encrypted does not make a difference so far.

igrr, 2015-11-30:
Using `esptool.py`:
```bash
esptool.py -p /dev/ttyUSB0 erase_flash
```

Or from inside the sketch:
```c++
ESP.eraseConfig();
```

ibaranov-cp, 2015-11-30:
Hello all,

Here is a compressed wireshark capture of trying to connect to esp in AP mode.
https://mrrobot.ca/owncloud/index.php/s/AKOX4BBZvoma47z

Actions taken
- Change WifiAccessPoint sketch to AP name ESP1, no password
- Connect to AP, try to hit 192.168.4.1
- Manually configure my IP to 192.168.4.10, Bcast 255.255.255.0, gateway 192.168.4.1
- Ping 192.168.4.1

I still get no connections.

ibaranov-cp, 2015-11-30:
Adding this to the example script has no effect:

void setup() {
  ESP.eraseConfig();
... etc

ibaranov-cp, 2015-11-30:
P.S
By default, my windows 7 laptop seems to assign IP 169.254.xx.xx with bcast 255.255.0.0
This usually means no good connection established.

ibaranov-cp, 2015-11-30:
To be clear, using a password results in no packet capture, and no successful connection.
Without a password, I get a connection and packets, but no IP assignment, and no usable connection.

luc-github, 2015-11-30:
did you tried to change the PHY mode on ESP ?

ibaranov-cp, 2015-11-30:
Adding WiFi.mode(WIFI_AP); either before or after the WiFi.softAP(ssid, password); statement has no effect.

The chip is never frozen, it continues to output serial data. etc. Just refuses to connect.

ibaranov-cp, 2015-12-04:
Hello,

(Updated to stable Nov 30th version just to be sure)

Two questions:
 What do the fields below indicate?
How can I do the phy mode set? I tried to hack it in myself, but no luck. Has anyone done it?

Here is the output of WiFi.printDiag(Serial);

Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
192.168.4.1

luc-github, 2015-12-05:
AP do not support phy mode N and seems you are in N, that may explain the connection issue.
I already gave the function above but here the extract from SDK for more details :
```
 wifi_set_phy_mode
Function:
Set ESP8266 physical mode (802.11b/g/n).
Note:
• ESP8266 soft-AP only support 802.11b/g.
• Users can set to be 802.11g mode for consumption.
Prototype:
bool wifi_set_phy_mode(enum phy_mode mode)
Parameter:
enum phy_mode mode : physical mode
enum phy_mode {
 PHY_MODE_11B = 1,
 PHY_MODE_11G = 2,
 PHY_MODE_11N = 3
};
Return:
true : succeed
false : fail  

```

ibaranov-cp, 2015-12-05:
I have the default example code, why does it do N?

Can I change it?

luc-github, 2015-12-05:
it is like password or SSID, it is saved / cached
default example purpose is not about PHY mode, so it is not present

of course you can change, just add the command

ibaranov-cp, 2015-12-05:
There does not seem to be a command to change it that I can see....

luc-github, 2015-12-05:
did you added the header to access it ?
```
extern "C" {
#include "user_interface.h"
}
```

ibaranov-cp, 2015-12-07:
Thanks for the tip, it was a good theory, but changing to either B or G has no effect on success rate of connection.

In addition to that, remember that both myself and @comino had success with certain devices connection to the default example, which seems to be N.

comino, 2015-12-07:
I still have the same issue. I don’t believe its a simple issue like adding a few lines of code, since its working for 50% of my devices.  I tried a lot! I must be a more complex issue, maybe even with the SDK. I wait until we change to 1.5 (as said there were apparently some issues in the SDK) and if the issue consists I will investigate further.

comino, 2015-12-09:
Still unresolved and two more Laptops unable to connect

tablatronix, 2016-02-29:
Any progress on this or further suspicions ?

I have the same problem, eg. IOS will not join AP with error `Unable to Join this network` IF i have a stored config that fails to autoconnect.

Same config output as above, changing PHY_MODE makes no difference.

This only happens if i start a softap and have a saved client config and it fails to connect.

So basically a `  WiFi.begin("nonexisting","password")` before the mode(AP) switch over and softap() call reproduces this. no small test sketch to post yet.

If I `eraseConfig()` reboot, then it works again.

wifi info seems the same, so it must be some other setting or RF setup that's sticking around cause `printDiag` is not giving me any hints.

tablatronix, 2016-02-29:
I just pulled 7 days of commits. 
f28c5be..5ab63dc  master     -> origin/master

And so far seems to work now, guess I should have done that first.

tablatronix, 2016-02-29:
ok well my specific problem might be fixed now.

I had to explicitly disconnect

```
  WiFi.disconnect(true);
  WiFi.mode(WIFI_AP);
```

Which is odd.

tablatronix, 2016-03-01:
This issue sounds like #1624 with different symptoms.

tablatronix, 2016-03-01:
So here's my reproduction sketch, this condition occurs if you do not start ap mode fast enough or don't wait long enough for it to time out.  I can reproduce fairly easily with IOS, i guess it doesn't like some channels.

as far as `waitForConnectResult` goes, 
My guess is that scanmode is still async scanning channels when `wifi_station_get_connect_status` returns `STATION_NO_AP_FOUND` I see a rogue scandone popup, and if I delay about 2000ms after it works again.

But I am at my limit of debugging

```
/* Create a WiFi access point and provide a web server on it. */

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>

/* Set these to your desired credentials. */
const char *ssid = "ESPap";
const char *password;

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
 * connected to this access point to see it.
 */
void handleRoot() {
	server.send(200, "text/html", "<h1>You are connected</h1>");
}

void setup() {
	delay(1000);
   
  Serial.begin(115200);
  Serial.setDebugOutput(true); 
  Serial.println();
  
  WiFi.setAutoConnect(false);
  
  // WiFi.begin("nonexisting","password"); // store
  WiFi.begin(); // restore

   /* THESE BOTH BREAK AP JOINS FOR SOME REASON , they give time for the reconnect to start scanning if my guess ( you will see a rogue async "scandone" */
  // Serial.println("connect result: " + (String)WiFi.waitForConnectResult());
  delay(6000);
    
  WiFi.printDiag(Serial);
  WiFi.mode(WIFI_AP); // this stops the autoconnect
  WiFi.printDiag(Serial);
  
	Serial.println();
	Serial.print("Configuring access point...");
	/* You can remove the password parameter if you want the AP to be open. */
	WiFi.softAP(ssid, password); // stick a channel 1 in there, and it still fails...
  delay(300);
  WiFi.printDiag(Serial);

	IPAddress myIP = WiFi.softAPIP();
	Serial.print("AP IP address: ");
	Serial.println(myIP);
	server.on("/", handleRoot);
	server.begin();
	Serial.println("HTTP server started");
}

void loop() {
	server.handleClient();
}
```

tablatronix, 2016-03-02:
And of course it works fine on my other device... sigh

sunepedersen, 2016-03-14:
I had the same experience in areas with a lot of wifi networks i believe. I tried changing channels and setting WiFi.setOutputPower(20.5) and it seems to have improved the situation.

ibaranov-cp, 2016-03-15:
Interesting, what is the default output power?

Anyone know of any current vs. power measurements done? And power vs
distance?

-- 

Ilia Baranov | Engineering Manager, Research Solutions | Clearpath Robotics

1425 Strasburg Rd., Kitchener, Ontario N2R 1H2 Canada
*T: * +1 (800) 301-3863 x839

*E:* ibaranov@clearpathrobotics.com | *W:* www.clearpathrobotics.com

*Your Unmanned Experts **TM*

This electronic communication (including any attachments) is CONFIDENTIAL
AND LEGALLY PRIVILEGED. Any unauthorized use or disclosure is strictly
prohibited. If you are not the intended recipient, you are hereby notified
that any review, retransmission, conversion to hard copy, copying,
circulation or other use of this message and any attachments is strictly
prohibited. The recipient should check this email and any attachments for
the presence of viruses. The company accepts no liability for any damage
caused by any virus transmitted by this email. Thank you.

tablatronix, 2016-03-15:
I also test in an area with lots of aps, 23 or so.

Vzor-, 2016-04-09:
I can confirm this is an issue with the esp sdk and not the arduino implementation, and can be fixed by adding  wifi_station_disconnect() before setting a new mode.

CaptClaude, 2016-05-22:
I am new to this and am trying to RTFM before posting.  Using ESPlorer and lua, I am experiencing the same difficulty with getting consistent connection to an iOS device (iPhone 5) when my ESP-12EV2 is in SoftAP mode. I have no trouble consistently connecting with OS X or Windows. I have been through so many iterations and changes that I have lost track of the threads I have been following and cannot give a concise Step A, Step B, Step C... account of my debug attempts (I must be ADD).  Suffice it to say that I am flashing "write_flash -fm dio 0x00000 ../nodemcu_integer_0.9.6-dev_20150704.bin 0x3fc000 ../esp_init_data_default.bin" when necessary and trying different orders of operations of setting up the SoftAP mode. None of the different SoftAP setup sequences allows me to connect my iPhone more than once unless I power off the ESP and turn WiFi off and back on on the iPhone. Once I power cycle the ESP and turn the iPhone WiFi on and off, I can pretty much switch between my home AP and my ESP without dropping the connection.  The following sequence in init.lua seems to be working (consistently for over 20min):
'

wifi.setmode(wifi.STATION)

wifi.sta.disconnect() -- this is supposed to fix the iPhone issue
print("Start soft AP as mizefi with pw=12345678 and ip=192.168.4.1")
wifi.setmode(wifi.SOFTAP) 

wifi.setphymode(wifi.PHYMODE_G)
local cfg={
	ssid="mizefi",
	pwd="12345678"
}

wifi.ap.config(cfg)
cfg={}
cfg ={
    ip="192.168.4.1",
    netmask="255.255.255.0",
    gateway="192.168.4.1",
	beacon="33"
}
wifi.ap.setip(cfg)
'

YannikW, 2016-08-14:
It seems I've a similar issue..
I configured my ESP as AP and connection works fine with my android phone.. (Wifi and TCP server works), but when I trie to connect to the Wifi with my both windows pcs it wont connect.. Somestimes it workes, but it lose connection after a couple of seconds.. And that few times the Wifi was working I was not always able to connect to my TCP server (maybe the wifi crashed but windows delayed the message)..

I tried all the fixes mentioned above but nothing solved the problem..

`//Start Access Point
  ESP.eraseConfig();
  wifi_set_phy_mode(PHY_MODE_11G);
  WiFi.setAutoConnect(false);
  WiFi.disconnect(true);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);

  //Start Server
  server.begin();
  server.setNoDelay(true);`

Anyone has a idea to fix this? It's stanged it workes on my phone every time, but not reliable on my computers..

rvireday, 2016-09-06:
I can offer one clue.   When I have my laptop connected to a physical port on the WAP, I can talk to the ESP device no problem.  Web service works.

When I switch the laptop to WiFi only, rnet::ERR_CONNECTION_REFUSED.

So physical to WAP, WiFi to ESP: good.    WiFi to WAP, WiFi to ESP: no good.

Nothing shows on the ESP console that it is even having a node try and connect.
-r

santon2000, 2016-09-22:
Hi, I also faced with this problem. 

Any news?

tablatronix, 2016-09-22:
You guys are probably gonna have to post info what versions what boards and minimal reporduction sketches. No one can really debug this but id be glad to look at it to help. Have you tested latest dev code for example.

ArGaspar, 2016-11-27:
@igrr my sketch is working excellent on esp-01 but on esp-12e I have to use ESP.eraseConfig(); always, the first time I connect to wifi, everything goes well but when I try for second time I cant so I have to use ESP.eraseConfig(); . any idea what can be causing this on esp-12e?

best regards

devyte, 2017-10-19:
This is a known issue. You have to erase the flash before uploading the sketch, or call eraseConfig(), or call WiFi.disconnect().
This is caused by the STA and SOFTAP interfaces sharing a single radio. When the station side tries to reconnect, or even search, it channel hops, and drags the SOFTAP side along for the ride. The client that's connected to the SOFTAP sees the ap drop out, has to search for it, connect, get DHCP, etc all over again, then the TCP connection needs to retry, etc.
The only workaround is to force the SOFTAP to the same channel as the AP that the STA is trying to connect to. If done correctly (i.e.: connect STA first, then get the channel, then start the SOFTAP on that channel), it should be ok.
Different laptops/cells/other devices, as well as different OSs, behave differently when faced with an AP that drops out of existence.
Closing.

closed 2017-10-19 by devyte

ThomasBj, 2018-01-29:
So what is the solution to this issue? It sounds like ESP8266 is not up to the task of acting as an access point.
Can somebody who has got it working please submit some code.

vks007, 2018-04-01:
I had the same issue as what others describe. I could mostly connect from my iPhone but very rarely from my windows PC. 
As suggested on this post, I erased the config on the ESP once by calling ESP.eraseConfig(); and then things were fine. The connection is also very very fast now. It connects in a jiffy!
I still have to see how to incorporate it in the sketch.

davidverweij, 2018-06-04:
The addition by @tablatronix (see below) has fixed my issues for using a iphone to connect to the AP. Though I did change WIFI_AP to WIFI_STA. I am using WifiManager and this was a suggestion for an older SDK.

> ok well my specific problem might be fixed now. I had to explicitly disconnect before turning on AP.   `WiFi.disconnect(true);`
`WiFi.mode(WIFI_AP);`  
Which is odd.

My snippet:

```
WiFiManager wifiManager;
WiFi.disconnect(true);
WiFi.mode(WIFI_STA);
if (!wifiManager.autoConnect(SSID,PASS)) {
   Serial.println("failed to connect and hit timeout");
   delay(3000);
   //reset and try again, or maybe put it to deep sleep
   ESP.reset();
   delay(5000);
}

```