esp8266/Arduino #1095: WiFiWebServer keeps resetting D1

aloz77, 2015-11-27:
I'm trying my new Wemos D1 with the latest esp8266 libraries on IDE 1.6.5. I compiled WiFiWebServer.ino from the examples. I get easily connected but unfortunately after a couple of seconds it keeps doing wdt resets with or without exception. It looks like this on the serial monitor:

```
Connecting to MOCKBA HE CAXAP
.....
WiFi connected
Server started
192.168.178.24

Exception (0):
epc1=0x40202d47 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffeeaf8 depc=0x00000000

ctx: sys 
sp: 3ffffe10 end: 3fffffb0 offset: 01a0

>>>stack>>>
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Connecting to MOCKBA HE CAXAP
.....
WiFi connected
Server started
192.168.178.24

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
...
```

Looks like the issue #431 but it was resolved as I understood. Any ideas?

comino, 2015-11-27:
HttpServer works just fine for me with latest version - can u post the relevant code?

aloz77, 2015-11-28:
I tried it with minimal wifi sketch like this:

```
void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  WiFi.begin(WLAN_SSID, WLAN_PASS);
}

void loop () {
}
```

wdt reboots after each couple of seconds. After adding delay(1); in the loop() however everything works stable. May be wifi code need some more attention after each loop() run?

Some delay also helped to make WifiWebserver.ino running fine:

```
  WiFiClient client = server.available();
  if (!client) {
    delay(10);  // Stability issues withot this delay()
    return;
  }
```

igrr, 2015-11-30:
This might be something specific to D1. I'm certainly not seeing this behaviour on other boards.

cow77, 2015-11-30:
At least not wrong with all D1 boards: D1 board works well with Arduino 1.6.7 Hourly Build 2015/11/24 04:49 and esp8266-2.0.0-rc2.

wemos, 2015-12-20:
@aloz77
where d1 board do you buy from?

aloz77, 2015-12-20:
http://www.aliexpress.com/item/Free-Shipping-Smart-Electronics-ESP-12E-WeMos-D1-WiFi-uno-based-ESP8266-shield-for-arduino-Compatible/32494825633.html

closed 2015-12-20 by aloz77

VK-sophie, 2015-12-20:
http://www.aliexpress.com/item/Advanced-Electronic-Module-for-uno-Kit-with-Button-LED-Sensor-RF-Servo-LCD-Makerduino-Better-Than/32498392093.html?spm=2114.01020208.8.17.IE6nZc

otuk, 2015-12-21:
is there a place to discuss  about "D1 wemos" arduino like esp8266 based board?   I run a pretty fast mac pro, never had issues with arduino ide before, until I installed the hardware/esp8266com directories and did  the setup as described in the wemos site.   I can run the basic examples.  BUT  the compile times and upload times are not believable, I am talking about minutes to upload, and close to a minute to compile, basic examples.  It does run, but need some direction why the compile and upload would be taking so long.  I noticed warnings turned on generate many warnings in ide for the esp8266 libarary code, but that still does not explain the upload time.

aloz77, 2015-12-21:
I had to downgrade to Arduino 1.6.5 in order to compile without errors/warnings.

Watch out for Tools -> Upload Speed setting in order to speed up the upload.

otuk, 2015-12-21:
@aloz77: Thx,  I was on 1.6.5 it was slow, I upgraded to 1.6.7 it is still slow, I turned off all warnings.  Also the recommended d1 wemos upload speed (921K) never worked , i consistently received esp_send errors during upload with that speed on macosx.   I tried other speeds and I can consistently upload with 230K.  I am trying to see if this is particular to my board or there are known issues with D1 wemos.  My ch340g driver was also newly installed from the link on wemos site.

aloz77, 2015-12-22:
Uploading via USB from Win7 with 921K to Wemos D1 no problem here. Takes 4-5 sec.

ericcastro, 2016-03-06:
otuk: I too had countless problems with the wemos d1 on mac... i had to switch to windows and most problems were gone. esp8266 support on the mac is very, very poor at the moment. I also think running Arduino on El Capitan only makes it even worse.

otuk, 2016-03-06:
Yes I solved mine by moving to linux as well, not doing much on mac, I did finish my mini project to wifi control lights by doing that  :)

Vamshi253, 2016-12-20:
Hi,

At the moment, I am engaged in developing a prototype of temperature and humidity monitoring device based on wemos D1 mini. 

So whenever the control is encountering client.connect() function or client.print() function, the program is crashing. It's indicating a exception (9) as follows:

Exception (9):
epc1=0x40203e77 epc2=0x00000000 epc3=0x00000000 excvaddr=0x2c38345a depc=0x00000000

ctx: cont 
sp: 3fff1980 end: 3fff1b80 offset: 01a0

>>>stack>>>
3fff1b20:  3ffe8a78 00001e62 3fff08cc 40203e71  
3fff1b30:  3ffe88ec 3fff0b4c 3ffe8cd0 1814436f  
3fff1b40:  3ffe88ec 3ffeec8c 3fff0a90 402072b4  
3fff1b50:  00000000 3ffe8354 3fff0a90 40203583  
3fff1b60:  3fffdad0 00000000 3fff0b44 40208a04  
3fff1b70:  feefeffe feefeffe 3fff0b60 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

I used esp exception decoder to decode the above exception. The result is as follows:

0x40203e77: WiFiClient::connect(char const*, unsigned short) at C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40203e71: WiFiClient::connect(char const*, unsigned short) at C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x402072b4: Print::println(char const*) at C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x40203583: loop at C:\Users\hp\Desktop\temptest/temptest.ino line 441
0x40208a04: loop_wrapper at C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

could anybody of you help me regarding this.

Thank you

yang550831, 2017-03-11:
change=> #include <WiFi.h>
write =>#include <ESP8266WiFi.h>
..
.
.
.
.
WiFi.begin(ssid, password); //line befor "while (...)....."
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);Serial.print(".");
  }