esp8266/Arduino #1097: Library Manager update overrides bundled esp8266 libraries

per1234, 2015-11-27:
If the Arduino IDE Library Manager is used to install a library that is also bundled with esp8266 then that library is installed to the sketchbook libraries folder and thus will be used in preference to the esp8266 bundled library. This issue is reported in the Arduino IDE repository: https://github.com/arduino/Arduino/issues/4064 but has not received much support for being fixed from the developers so any input there might help to give it priority. I made a list of workarounds for this issue here: https://github.com/JChristensen/mighty-1284p/issues/24

igrr, 2016-02-29:
Closing this as https://github.com/esp8266/Arduino/commit/f2949c1ee12a9cfee182db3700e02cecdaf5d601 has been merged and released.

closed 2016-02-29 by igrr

per1234, 2016-03-01:
https://github.com/esp8266/Arduino/commit/f2949c1ee12a9cfee182db3700e02cecdaf5d601 doesn't fix this issue, it only reduces the chance of it happening by preventing updates cause by false updatable library notifications. But if there is actually an update for any of the libraries included with the Arduino IDE in the {IDE install folder}/libraries folder then this issue will still occur. I don't think there is any good solution for completely fixing the issue in this repository but it could be helpful to add a note about it in the documentation. Something like: https://github.com/JChristensen/mighty-1284p/commit/ceed9b68cc913d8465ef10ccb59d6bbc14b870b6. I at least wanted to bring attention to this issue as a likely cause of future issue reports in this repository.

The approach that the Arduino 101 core is using to deal with this issue is to attempt to add support for their hardware to all the relevant libraries and then removing their copies from the package when that is accomplished. Now of course it seems unlikely that the Arduino developers would be willing to merge changes to support ESP8266 but I was surprised to see a [PR](https://github.com/arduino/Arduino/pull/4547) recently merged to add support for ATmega1284/ATmega1284P(not used in any official Arduino board) to the Servo library.