esp8266/Arduino #1098: system_get_rst_info now returns always "external system reset" after deep-sleep and cold-boot

flyinggorilla, 2015-11-28:
this worked fine in the stable release with v1.6.5. returned 0 on power-on and 5 on deep-sleep-wakeup.

however, now since i updated to 2.0.0-rc1/rc2 and included SDK1.3.0 i get always return value 6 instead of the proper one. 

using a Huzzah board, Arduino IDE 1.6.5-r5.
```
extern "C" {
  #include "user_interface.h"
}

// returns -1 on error
int getResetReason() {
  rst_info* ri = system_get_rst_info();
  if (ri == NULL)  
    return -1;

  return ri->reason;
} 

void setup() {
  
  int resetReason = getResetReason();
........
```

igrr, 2015-11-28:
This is a known bug in SDK 1.3, see http://bbs.espressif.com/viewtopic.php?f=7&t=992&p=3372.
There is a fixed replacement libmain provided by Espressif, but it is known to have other issues.
This will be fixed automatically when we switch to SDK 1.4.

closed 2015-11-28 by igrr

chaeplin, 2015-11-28:
SDK 1.5 is released.

http://bbs.espressif.com/viewtopic.php?f=7&t=1230 : system_get_rst_info() odd behaviour in SDK1.4
http://bbs.espressif.com/viewtopic.php?f=46&p=4795 : [SDK Release] ESP8266_NONOS_SDK_V1.5.0_15_11_27

Resolved Issues(Bugs below are eligible for Bug Bounty Program):
Returned value of system_get_rst_info is wrong.

Added APIs:
espconn_abort: break TCP connection compulsively.
espconn_secure_delete: delete the SSL server when ESP8266 runs as SSL server.
wifi_station_set_cert_key: set certificate and private key for WPA2 Enterprise.
wifi_station_clear_cert_key: release the resouce of connecting to WPA2 Enterprise AP, and clear the status.