esp8266/Arduino #1101: Commit 4f35207 crashes my NodeMCU

skorokithakis, 2015-11-28:
I'm trying to flash an image to my NodeMCU v1.0 and it resets at the latest head. Bisecting shows this commit as being the culprit:

~~~
4f352079519e549722e9cad03c2e268a8e5c6277 is the first bad commit
commit 4f352079519e549722e9cad03c2e268a8e5c6277
Author: Ivan Grokhotkov <igrokhotkov@gmail.com>
Date:   Tue Oct 6 16:11:40 2015 +0300

    Move some core functions back into RAM

:040000 040000 b286ecd5204aeb4b6024817db295bce0d3660042 aab25438727025d899a69199e86bf036c25984e9 M	cores
~~~

Output from a bad commit:

~~~
Terminal ready

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
~~~

igrr, 2015-11-28:
Could you please provide the sketch which triggers this issue?

skorokithakis, 2015-11-28:
Sure:

~~~
void setup() {
    Serial.begin(115200);
}

void loop() {
    Serial.println("test");
    delay(1000);
}
~~~

skorokithakis, 2015-11-28:
I run it as:

~~~
platformio run; picocom /dev/ttyUSB0 --baud 115200
~~~

igrr, 2015-11-28:
I tried this sketch with latest version and it works fine with NodeMCU 1.0.

I'm not familiar with platformio, but a quick look at their build scripts
shows that platformio comes with its own set of linker scripts, which are
not compatible with the changes you linked to.

Please try building the same sketch with Arduino IDE (1.6.5) instead and
let me know if this solves your issue.

skorokithakis, 2015-11-28:
Unfortunately I spent two days trying to get a sketch working, and finally settled on the Arduino IDE's version of this repo, which works. I'm afraid that if I remove that to go back to HEAD to test this, it will stop working. This might very well be a PlatformIO linker issue, so feel free to close this issue. I'll open one to PlatformIO to change their linkers.

closed 2016-06-23 by igrr