esp8266/Arduino #1102: SDK 1.5 is released.

chaeplin, 2015-11-28:
http://bbs.espressif.com/viewtopic.php?f=46&t=1442

WPA2 Enterprise!!!

Release Notes
Resolved Issues(Bugs below are eligible for Bug Bounty Program):
Returned value of system_get_rst_info is wrong.

Optimization: 
Optimize espconn_regist_recvcb for UDP transmission.
Print information will prompt errors if the input parameters of os_time_arm excess the upper limit.
Optimize memory management.
Optimize RF frequency offset.
Print information will prompt errors if the size of bin file is too large in OTA.
Optimize mDNS.
Revise UART output print error when powered on again after call system_uart_swap.
Revise errors during multiple UDP transmissions.
Revise the error when wifi_station_set_config_default is called and the same AP is set as before, while information cannot be stored in the Flash.
Optimize the error of packet loss during UDP transmission process under single station mode.
Add new function, WPA2-Enterprise is supported.
SmartConfig version is upgraded to V2.5.3.
Mesh version is upgraded to V0.2.3.
User application needs to add "-lcrypto" in 'LINKFLAGS_eagle.app.v6" area of Makefile.

Added APIs:
espconn_abort: break TCP connection compulsively.
espconn_secure_delete: delete the SSL server when ESP8266 runs as SSL server.
wifi_station_set_cert_key: set certificate and private key for WPA2 Enterprise.
wifi_station_clear_cert_key: release the resouce of connecting to WPA2 Enterprise AP, and clear the status.

alltheblinkythings, 2015-12-22:
FYI, there's a SDK patch for liblwip.a, fixes a bug in espconn_abort().  http://bbs.espressif.com/viewtopic.php?f=46&t=1527:

> [SDK Patch] espconn_abort issue on ESP8266_NONOS_SDK_V1.5.0
>
> Here is a patch based on ESP8266_NONOS_SDK_V1.5.0 solved a problem that calling espconn_abort may cause unexpected reset.

Don't think we use that API yet, but mentioning in case it becomes important later on.

blensonpaul, 2016-01-12:
Anyone with EAP/TTLS Radius authorisation example code? Thanks

joostd, 2016-01-12:
Version 1.5.1 of the SDK is released:
http://bbs.espressif.com/viewtopic.php?f=46&t=1609

For WPA2-Enterprise (EAP-TTLS) this is an important update, as libwpa2 now contains a function `wifi_station_set_username` that will set the username (that defaults to "Espressif").

halcy, 2016-02-03:
Have you had any luck with actually _using_ wifi_station_set_username ? It is there in the library, but I cannot seem to find it documented (or, really, exposed) anywhere.

blensonpaul, 2016-02-03:
please check thread in esp forum. there is some code. 

http://www.esp8266.com/viewtopic.php?f=6&t=1243&hilit=wpa2&start=20

joostd, 2016-02-03:
There is no documentation unfortunately. Use these functions as follows:

int wifi_station_set_username(u8 *username, int len); 
void wifi_station_clear_username(void);

halcy, 2016-02-03:
I guessed as much, though I seem to be unable to actually connect to a network (eduroam, incidentally) even with the username thusly set.

I might just be calling the function wrong, or might have something in the wrong state still (I tried to expose it to NodeMCUs lua interpreter). I'll likely prod around some more tomorrow.

DKfraDK, 2016-02-15:
@halcy did you manage to connect to eduroam? I am doing a project in a course at CS, and it would be really helpful if I get an ESP8266 connected to it.

joostd, 2016-02-15:
I have committed some sample code at https://github.com/joostd/esp8266-eduroam, but it needs some cleaning up - will do that later. Hopefully this will get you started. Note that for Arduino, you need some patches - see https://github.com/joostd/Arduino (I will create a PR).

halcy, 2016-02-15:
@DKfraDK I have not had any luck in getting things to work without requiring a client certificate, which we don't use here (PEAP). I do wonder if a DFN-signed certificate works for logging into eduroam...

joostd, 2016-02-15:
Logging into eduroam with a client certificate only works if your eduroam home institution (the Identity Provider) supports eap-tls. If only PEAP is supported, I guess you will have to wait until the espressif SDK implements it.

DKfraDK, 2016-02-15:
Thanks for the fast replies guys. Im not sure if my university support eap-tls, so far I have only used PEAP. So I guess i'll try to contact the IT-department and see if I can get a client certificate.

joostd, 2016-02-15:
@DKfraDK you may want to check https://cat.eduroam.org to see if your university is listed.

closed 2016-02-27 by igrr