esp8266/Arduino #1103: softAP(ssid, password) ignores new data and uses old one

starlino, 2015-11-28:
I have weird problem I am basically running this sample witch current rc2.0.0-rc2:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

I have previously used this board with other scripts and somehow the ESP remembers the old SSID/Pass for access point, and ignores the new one

  WiFi.softAP(ssid, password);

I tried   WiFi.persistent(false); but it did not help. 

If I do:   Serial.setDebugOutput(true);  

I get:

Configuring access point...
softap config unchanged
AP IP address: 192.168.4.1
HTTP server started
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

So why "softap config unchanged"  ?

starlino, 2015-11-28:
An update:  switching to  rc2.0.0-rc1  fixed the issue above, there might be something wrong in  rc2.0.0-rc2,

igrr, 2015-11-28:
Duplicate of #1054. Fixed in git now.

closed 2015-11-28 by igrr

sayanee, 2017-07-28:
@igrr @starlino I have exactly the same problem, but I'm unable to get Arduino to use the latest [rc1](https://github.com/esp8266/Arduino/releases/tag/2.4.0-rc1) version.

I tried following the [readme for using Arduino with git version](https://github.com/esp8266/Arduino#using-git-version), but installing in the directory did not seem to work `~/Documents/Arduino/hardware/esp8266com`. After I re-flashed ESP8266, the SSID for the access point is still the old name.

sayanee, 2017-07-28:
Ok, I got the `2.4.0-rc1` loaded into Arduino via the Boards Manager by appending `https://github.com/esp8266/Arduino/releases/download/2.4.0-rc1/package_esp8266com_index.json`:

[image]

But the AP still shows the old one even after re-flashing 🤔

sayanee, 2017-07-28:
Alright - I have gotten it to work with the [example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino) 👍

guruthree, 2019-11-22:
Possibly not for the same reasons, but I had an issue that presented the same way: changing the configured SSID/password didn't take effect. It appears that this is what happens when your password isn't at least 8 characters long. Hopefully this keeps someone else from spending a few hours tracking this down...

fgerard, 2020-09-09:
Thanks!! The thing is as @guruthree says, if password is not 8 chars or more everything is ignored!!

Bagunceiro, 2022-04-23:
Good grief. That's a couple of hours of my life I won't get back. Is there a fix in the works for that?

bobemoe, 2023-02-16:
Wow what a weird bug! Big thanks @guruthree for saving me a few hours :D

Himanshu21358, 2023-03-10:
Sir I am new developer now we are create a ESP8266 as server and serial port receive data and send data successfully. but I have a issue now I am change my Wi-Fi SOFTAP credentials(like SSID, PASWORD) by receive in serial monitor, so please suggest any code ?