esp8266/Arduino #1107: OTA updates fail

ktallevi, 2015-11-29:
Testing out sketch: ESP8266WebServer/examples/WebUpdate/WebUpdate.ino

On initial upload of a simple sketch that prints hello world, the OTA update fails and the ESP8266 hangs. Cycling the power and uploading the sketch again seems to work. Any ideas why the first attempt always fails? Log below:

Booting Sketch...
Ready! Open http://esp8266-webupdate.local in your browser
Update: ESP8266PCB.cpp.generic.bin
sleep disable
Update Success: 266384
Rebooting...

 ets Jan  8 2013,rst cause:1, boot mode:(1,0)

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset

**cycle power**

Booting Sketch...
Ready! Open http://esp8266-webupdate.local in your browser
Update: ESP8266PCB.cpp.generic.bin
sleep disable
Update Success: 266384
Rebooting...

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld

**new sketch starts up OK**

Links2004, 2015-11-29:
i think its the same root cause then #1017

reaper7, 2015-11-29:
I have the same problem (nodemcu v1.0)

ktallevi, 2015-11-29:
I'm using the Olimex MOD-WIFI-ESP8266-DEV, which uses 2K pullups on GPIO0 instead of 10K. I flashed a simple sketch (no OTA) which waits 2 seconds and issues ESP.restart();

 ets Jan  8 2013,rst cause:1, boot mode:(1,0)

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset

Cycling the power, my simple restart sketch runs as expected, restarting without issue many times over. 

Any ideas? 

Thanks.

ktallevi, 2015-11-29:
Update: If I do a physical power cycle after flashing a sketch (OTA and non-OTA), the ESP will restart OK.

igrr, 2015-11-29:
Let's move the discussion to #1017, which is basically the same issue.

closed 2015-11-29 by igrr