esp8266/Arduino #1109: File uploads in FSBrowser.ino example

AdySan, 2015-11-29:
Question for @me-no-dev 

Thanks a lot for the filesystem browser example here: https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/FSBrowser

The edit.htm seems to be some packed format html, so it's not really readable. I'm trying to make a simple drag and drop interface for uploading .html and .js files to the SPIFFS. I tried using [this](http://hayageek.com/drag-and-drop-file-upload-jquery/) but the POST request goes to server.onNotFound() for some reason. Any pointers would be very helpful.

me-no-dev, 2015-11-30:
what is the upload URL that you use?

me-no-dev, 2015-11-30:
Not sure if it's the same exact version as in the example but here is the source that is in the SD Browser example: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/SDWebServer/SdRoot/edit/index.htm

AdySan, 2015-11-30:
`var uploadURL ="http://adyesp8266.local/edit"; //Upload URL`

Actually, it seems to be working, I do get a `LmacRxBlk:1` once in a while, but the upload is successful.  Not sure what was going wrong. Let me finish it up and I will post details here.

And yes, I just stumbled upon that file, it is very similar indeed!

AdySan, 2015-11-30:
Here's the whole project on GitHub: https://github.com/AdySan/ESPSocket

Still a lot of rough edges here and there but the basics are working. Thanks for making the awesome examples, they're extremely helpful!

me-no-dev, 2015-11-30:
nice :)

danielmawhirter, 2015-11-30:
Much appreciated, I was about to ask this question myself, as I am using parts of it toward controlling some costume hardware. The only major version difference appears to be directory support, since the SD supports it and SPIFFS does not.

closed 2015-12-01 by AdySan

mkeyno, 2015-12-21:
hi guys
can anyone share the correct links for each of handler ? I've tried couple of links but it seems just on handler called