esp8266/Arduino #1118: Version 2.0.0 is broken?

ntasher, 2015-11-30:
I have IDE 1.6.5 and the ESP8266 1.6.5-947 worked nicely for me. I just saw 2.0.0 was uploaded and I updated via board manager but ALAS I get the following error:
xtensa-lx106-elf-gcc: error: C:\Users\xxxxx\AppData\Local\Temp\build1631103587031967139.tmp/arduino.ar: No such file or directory
Error compiling.
Moving back to 1.6.5-947 and the project compiles just fine.
What's wrong?
Thx, Nir

asetyde, 2015-11-30:
I clean all folder , go to Program - app data - and find arduino 15 clean all package

ntasher, 2015-11-30:
Tried clearing the Arduin15 directory and reinstalling via board manager. Same error with "blink".

asetyde, 2015-11-30:
My work perfectly , maybe try new installing of ide

igrr, 2015-12-01:
@ntasher could you please enable verbose compiler output in Arduino preferences, and share the full build log via pastebin?

ntasher, 2015-12-01:
@igrr Here goes:
https://gist.github.com/ntasher/4ea655cd237b22b84674
Thanks!

igrr, 2015-12-01:
(accidentally replied to the gist)

That's weird. First it uses the correct path:
`C:\Users\ntasher\AppData\Local\Temp\build1256863664410631453.tmp/arduino.ar`
And then the final linker command has the wrong one:
`C:\Users\ntasher\AppData\Local\Temp\build1256863664410631453.tmp\core/arduino.ar`
I have downloaded 1.6.5-r5 (OS X) and it doesn't seem to have this issue. Perhaps this is an issue with 1.6.5-r2 or 1.6.5-r3? I can't find download links for these older versions on Arduino website anymore to verify this, unfortunately.
Could you please try upgrading the IDE to the latest 1.6.5?

ntasher, 2015-12-01:
OK. Managed to find the 1.6.5-r5 and now everything seems to compile OK.
Now it's time to try SPIFFS :-)

igrr, 2015-12-01:
For the record, which 1.6.5 release did you use before?

closed 2015-12-01 by igrr

ntasher, 2015-12-01:
I can't tell which version. It must have been the original release of 1.6.5.-------- Original Message -------- From: Ivan Grokhotkov  Sent: Tue, Dec 1, 2015 09:36 To: esp8266/Arduino  CC: ntasher  Subject: Re: [Arduino] Version 2.0.0 is broken? (#1118)For the record, which 1.6.5 release did you use before?