esp8266/Arduino #1120: OTA headers clarification

marvinroger, 2015-11-30:
I am implementing OTA server-side, and it's not that clear what `Flash chip size is 2x the size of the sketch` condition means. Is it 2x the size of the current sketch, or the size of the new to-be-sent sketch? I though the condition would be `newSketchSize + oldSketchSize <= flashSize`... No?

Moreover, we have the following size-related headers, (from the doc): 

* `HTTP_X_ESP8266_CHIP_SIZE` = 524288
* `HTTP_X_ESP8266_SKETCH_SIZE` = 373940
* `HTTP_X_ESP8266_FREE_SPACE` = 671744

So here, the update is impossible, since `HTTP_X_ESP8266_CHIP_SIZE` is not 2x `HTTP_X_ESP8266_SKETCH_SIZE`. If this is right, this should be updated in the docs.

How can `HTTP_X_ESP8266_FREE_SPACE` be greater than `HTTP_X_ESP8266_CHIP_SIZE`? Data is not coherant. Or I'm lost and the does not sufficiently guide.

Thanks!

Links2004, 2015-11-30:
HTTP_X_ESP8266_SKETCH_SIZE  - is the current size of the sketch on the flash chip
HTTP_X_ESP8266_CHIP_SIZE - is the real size of your flash ship
HTTP_X_ESP8266_FREE_SPACE - is what the software think its free

the free space is most likely wrong because you select the wrong size in the IDE. 
try: https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino to check the config

on the server you can check if the free_space is more or the same then the new binary.
If you not check it and simply send the file, the code on the ESP will detect it and automatically about the update.

```Flash chip size is 2x the size of the sketch```
means if you habe a sketch that needs 256KB for example,
than you need at minimum a 512KB Flash chip to be able to use OTA updates.

marvinroger, 2015-11-30:
Yes, the free space is wrong, but it is wrong [in the docs](http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html#http-server) (the data above is taken straight from the docs, this is not my data), so it needs to be updated there as it is confusing.

And I know that `Flash chip size is 2x the size of the sketch` means if the sketch is 256KB then 512KB flash chip is needed, but this assumption is wrong, because a sketch will see its size change after updates.
Imagine you have a 512KB chip size, and the current sketch is 257KB. You could have lighten your sketch with some optimization, and the newer version might be 250KB, so it would fit, even if the chip size was not two times the one of the initial sketch.

Do you know what I mean? This would deserve to be fixed in the doc, don't you think?

Links2004, 2015-12-01:
shut now be better to understand.

marvinroger, 2015-12-01:
Yes, thanks!

closed 2015-12-01 by marvinroger