esp8266/Arduino #1124: 2.0.0-rc2, comm led blinks so fast and hot

chaeplin, 2015-12-01:
With blank sketch / ESP-12

Arduino/IDE 1.6.5 / Boards Manager 2.0.0-rc2
--> comm led :  blink so fast --> heat up / hot

Arduino/IDE 1.6.5 / Boards Manager 2.0.0-rc1
--> comm led : no blink

igrr, 2015-12-01:
Could you please share the .elf output file for your 2.0.0-rc2 test case?

chaeplin, 2015-12-01:
Here : https://github.com/chaeplin/esp8266_and_arduino/raw/master/temp/sketch_dec01a.zip

igrr, 2015-12-01:
I have flashed your .bin file to a nodemcu 1.0 board, and the output looks fine:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```
UART LED isn't blinking.

Perhaps you could try erasing flash?
```
esptool.py -p /dev/tty.SLAB_USBtoUART erase_flash
```

chaeplin, 2015-12-01:
I have tested more.

* hot -- > one of my chip was damaged.
* blink so fast ---> not blink.

Looks like serial debug out(no Serial.begin) with esp8266 test board / Generic module(nothing connected to uart)

* sketch
https://gist.github.com/chaeplin/4683dc2540735a3ade1f

* cam
https://github.com/chaeplin/esp8266_and_arduino/blob/master/temp/IMG_5459.MOV?raw=true

closed 2015-12-01 by chaeplin