esp8266/Arduino #1133: Can't control a servo once detach()ed (2.0.0-rc2)

kotobuki, 2015-12-03:
The following code works as expected with my Arduino Uno board, but doesn't work with my ESP8266 based board. Once a servo is detached, the code can't control the servo anymore. According to the Library Manager, the version of the Servo library is 1.1.0.

```
#include <Servo.h>

Servo servo;

const int servoPin = 2;

void setup() {
  servo.attach(servoPin);
  servo.write(90);
  delay(500);
  servo.detach();
  delay(500);
}

void loop() {
  servo.attach(servoPin);

  servo.write(60);
  delay(500);
  servo.write(120);
  delay(500);
  servo.write(90);

  servo.detach();
  delay(500);
}
```

drewfish, 2015-12-03:
I have the same problem. I want to detach while the device is asleep.

igrr, 2015-12-05:
/cc @Makuna

Makuna, 2015-12-08:
I will investigate.
please confirm if you are using Stable or Staging versions.

kotobuki, 2015-12-16:
I tried with 2.0.0 rc 2 (i.e. Staging) and 2.0.0 (i.e. Stable at current moment). The library manager says the Servo library is updatable, but I couldn't update to 1.1.1 (failed).

[image]

Makuna, 2015-12-16:
@kotobuki I am unsure if the servo library you are references above is even supported on ESP.  

The Servo library that works with ESP is part of the ESP Arduino package.

Thanks, I will be looking at it today.

Makuna, 2015-12-16:
I have a fix and will get a pull out for it as soon as I can slay this GIT dragon that is stopping me from publishing a branch on my fork of this depot.

closed 2016-02-27 by igrr

geekmustafa, 2017-09-29:
pro us a time after detach and use this myservo.attach(9);
because the detach is stop servo and myservo.attach(9); 
simple its like switch ^_^

Makuna, 2017-09-29:
@geekmustafa Commenting on a closed issue is not a good practice.  If there is an issue you are having that is related to this, create a new issue and reference this one.  General questions are best asked on the gitter channel.