esp8266/Arduino #1134: UDP broadcast addresses don't work

vogtitec, 2015-12-03:
I am using the ESP for receiving Artnet messages. For directed messages everything works fine, but it seems that broadcast is completely "ignored" by the ESP8266...
If the module has e.g. a localIP of 192.168.1.25, it doesn't receive messages with an IP of 192.168.1.255 (which is used for Artnet Poll Requests).
On the other hand, the ESP doesn't send any packet, that is directed to a broadcast address, although the return values of udp.beginPacket, udp.write and udp.endPacket all are "successful" (1)...
I sniffed with wireshark, but the ESP doesn't send anything, if I use "192.168.1.255" as ip.

Has anyone any idea how to fix that?

I am currently using ESP Version 2.0.0 rc-2.

Any thoughts on this are welcome, thank you!

me-no-dev, 2015-12-03:
what is the size of the broadcast packet that you want to send?
Recently the starting size of the UDP buffer was reduced and I experienced such things trying to broadcast packets larger than that initial buffer.

vogtitec, 2015-12-03:
The size is 239 bytes... but this wouldn't explain the missing reception of broadcast messages ?!

bbasil2012, 2015-12-03:
@vogtitec 
2.0.0-stable
Send UPD broadcast very well:

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
IPAddress broadcast_ip;
WiFiUDP sender_udp;

byte udp_buf[10]={0,1,2,3,4,5,6,7,8,9};
void setup() {
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) {
          delay(500);
          Serial.print(".");
  }
   broadcast_ip=WiFi.localIP();
   broadcast_ip[3]=255;
}

void loop() {
  yield();
  if (millis() % 1000==0) {
    sender_udp.beginPacket(broadcast_ip,8266);
    Serial.println("Send "+String(sender_udp.write(udp_buf,sizeof udp_buf))+" bytes");
    sender_udp.endPacket();
  }
 
}

me-no-dev, 2015-12-03:
no it would not :) there we have to dig in a different direction.
I have some code I have used to run multicasts on the ESP, let me see where it is...
Here it is, see if it will do if you set the multicast IP as 192.168.1.255

_Multicaster.h_
```cpp
#ifndef MULTICASTER_H
#define MULTICASTER_H

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

class UdpContext;

class Multicaster {
  protected:
    UdpContext* _server;
    uint32_t _multicast_addr;
    uint16_t _multicast_port;
    void (*_onPacketCallback)(uint8_t* data, size_t len, uint32_t from_ip, uint16_t from_port);
    void _onRx();

  public:
    Multicaster();
    ~Multicaster();
    
    bool begin(IPAddress multicats_ip, uint16_t port, uint32_t ttl=1);
    size_t write(uint8_t *data, size_t len);
    void send();
    void send(uint32_t address, uint16_t port);
    size_t send(uint8_t *data, size_t len, uint32_t address, uint16_t port);
    size_t send(uint8_t *data, size_t len);
    void onPacket(void (*cb)(uint8_t*, size_t, uint32_t, uint16_t)){
      _onPacketCallback = cb;
    }
};

#endif
```

_Multicaster.cpp_
```cpp
#define LWIP_OPEN_SRC
#include <functional>
#include "Multicaster.h"
#include "WiFiUdp.h"
#include "debug.h"

extern "C" {
  #include "osapi.h"
  #include "ets_sys.h"
  #include "user_interface.h"
}

#include "lwip/opt.h"
#include "lwip/udp.h"
#include "lwip/inet.h"
#include "lwip/igmp.h"
#include "lwip/mem.h"
#include "include/UdpContext.h"

Multicaster::Multicaster(): _multicast_port(0), _multicast_addr(0), _server(0){}

Multicaster::~Multicaster(){}

void Multicaster::_onRx(){
  while(_server->next()){
    uint32_t from = (uint32_t) _server->getRemoteAddress();
    uint16_t  port = _server->getRemotePort();
    size_t len = _server->getSize();
    uint8_t buffer[len];
    _server->read((char*)buffer, len);
    if(_onPacketCallback)
      _onPacketCallback(buffer, len, from, port);
  } 
}

bool Multicaster::begin(IPAddress multicast_ip, uint16_t port, uint32_t ttl){
  if (_server)
    return true;

  struct ip_info local_ip;
  int mode = wifi_get_opmode();
  if(mode & STATION_MODE){
    wifi_get_ip_info(STATION_IF, &local_ip);
  } else if (mode & SOFTAP_MODE) {
    wifi_get_ip_info(SOFTAP_IF, &local_ip);
  } else {
    return false;
  }
  
  _multicast_addr = (uint32_t) multicast_ip;
  _multicast_port = port;
  
  ip_addr_t maddr;
  maddr.addr = (uint32_t) multicast_ip;
  if (igmp_joingroup(&local_ip.ip, &maddr) != ERR_OK)
    return false;

  _server = new UdpContext;
  _server->ref();
  
  if (!_server->listen(*IP_ADDR_ANY, _multicast_port))
    return false;

  _server->setMulticastInterface(local_ip.ip);
  _server->setMulticastTTL(ttl);
  _server->onRx(std::bind(&Multicaster::_onRx, this));
  return _server->connect(maddr, _multicast_port);
}

size_t Multicaster::write(uint8_t *data, size_t len){
  _server->append(reinterpret_cast<const char*>(data), len);
  return len;
}

void Multicaster::send(){
  _server->send();
}

void Multicaster::send(uint32_t address, uint16_t port){
  ip_addr_t daddr;
  daddr.addr = address;
  _server->send(&daddr, port);
}

size_t Multicaster::send(uint8_t *data, size_t len){
  write(data, len);
  send();
  return len;
}

size_t Multicaster::send(uint8_t *data, size_t len, uint32_t address, uint16_t port){
  write(data, len);
  send(address, port);
  return len;
}
```

_Usage_
```cpp
include "Multicaster.h"
const IPAddress mip(192,168,1,255);//224.0.0.0 to 239.255.255.255
const uint16_t mport = 1234;
Multicaster server;
void onPacket(uint8_t *data, size_t len, uint32_t address, uint16_t port){
  //deal with packet data
}

void setup(){
  if(server.begin(mip, mport)){
    server.onPacket(onPacket);
  }
}

void loop(){
  char *out = "Hello World";
  server.send((uint8_t*)out, strlen(out));
  delay(1000);
}
```

me-no-dev, 2015-12-03:
if you end up using the above and it works, keep in mind that onPacket is called from network interrupt so do not spend time there :) maybe grab the data and deal with it in the loop

vogtitec, 2015-12-03:
@me-no-dev 
Thank you, but:
Trying your sketch leads to an exception at the point where "server.begin" is called with the broadcast ip...:
Starting ES8266
.......WiFi connected on 192.168.1.27
Trying to start server...

Exception (28):
epc1=0x40201b99 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000014 depc=0x00000000

ctx: cont 
sp: 3ffef8c0 end: 3ffefad0 offset: 01a0

>>>stack>>>
3ffefa60:  00000001 00000000 3ffefb1c 40202fac  
3ffefa70:  1b01a8c0 c801a8c0 3ffefb1c 3ffeeaa8  
3ffefa80:  3ffee9a0 3ffee980 3ffefb1c 40201de1  
3ffefa90:  3ffe86c8 3ffee980 0000000b c801a8c0  
3ffefaa0:  3ffe86c8 00000000 3ffeeaa0 40201e1a  
3ffefab0:  3fffdc20 00000000 3ffeeaa0 402024f5  
3ffefac0:  00000000 00000000 3ffeeab0 40100114  
<<<stack<<<

vogtitec, 2015-12-03:
@bbasil2012 
That's right, your example works indeed!
I noticed the following:
As long as WiFi is configured to use DHCP, broadcasting works!
But as soon as you configure WiFi to use a static IP:
       WiFi.config(config.IP,config.Gateway,config.Gateway,config.Netmask);
       WiFi.begin(config.ssid.c_str(), config.password.c_str());
broadcast addresses fail!

May this lead to any further findings??

me-no-dev, 2015-12-03:
@vogtitec maybe comment
```cpp
_server->setMulticastInterface(local_ip.ip);
_server->setMulticastTTL(ttl);
```
not sure about:
```cpp
if (igmp_joingroup(&local_ip.ip, &maddr) != ERR_OK)
    return false;
```
but might as well try it
one of those lines probably leads to the exception

vogtitec, 2015-12-03:
Oh well... I must have been too blind ... a look at ESP8266WiFi.h made it obvious:
I just missed the order of arguments taken by WiFi.config... the third argument has to be the netmask,
and I wrongly placed the gateway address in it...
I changed the config line to
WiFi.config(config.IP,config.Gateway,config.Netmask);
and everything is ok now!

Thanks to all for your help!!!

closed 2015-12-03 by vogtitec

neut2012, 2017-05-04:
The code @me-no-dev provided works like a charm with sming framework :+1: