esp8266/Arduino #1135: Multiple servos with different duty cycles

iainmckay, 2015-12-03:
I'm trying to get multiple ESC's initialized and running with different duty cycles. My sketch initializes all the ESC's but they run at the same speed. 

Looking through the source code these servos will be controlled by the same timer, is this likely the source of my issue?

You can view my sketch [here](https://gist.github.com/iainmckay/58be40d6d827d80aa2c8) and appreciate any help.

AndyKirby, 2015-12-03:
Apologies in advance for my reply, it is not directly answering your
question.

For real time control applications (particularly ones that use bit
banging) the ESP8266 is a bit like the Raspberry Pi in that there is a
lot of other background stuff going off at the same time and this has to
have a higher priority than the app you are adding. The servo control on
the ESP as I understand it is software rather than a dedicated hardware
peripheral and suffers because of this.

It is worth having a look at the servo controllers for the Raspberry Pi
and seeing how they have fixed this issue.

They use a dedicated PWM driver chip that will keep the servo signal
timings regular and jitter free.

https://www.adafruit.com/products/2327

I think there may be a similar breakout board for the arduino that you
could press into service, check aliexpress. Or better still mcguyver
yourself one up.

However I do understand that you might want to try and get it all out of
the ESP8266 or you may just be having an experiment with servo's.

Hope this helps

Cheers

Andy Kirby

PS the new ESP when it arrives being dual core will be much better in
this respect. Or at least should be.

iainmckay, 2015-12-03:
Thank you for the reply. For the moment it's just a bit of tinkering while I wait (the many weeks) for a bunch of chips to arrive for handling PWM.

Regards,

me-no-dev, 2015-12-03:
what do you mean by "the same speed"? The duty of them all is the same?
We do use software PWM but it is based on hardware interrupt timers so it is just about jitter free (I have a mini quad running on ESP).

me-no-dev, 2015-12-03:
This will let you use the PWM instead of the Servo driver to control ESCs/Servos. Report if this works and the other does not, so we can look into the code.
```cpp
//setup pwm for 50Hz (20ms)
analogWriteFreq(50);
//setup pwm for range of 20000 or 1 equals 1uS
analogWriteRange(20000);
//now set a servo on GPIO 12 to middle
analogWrite(12, 1500);
```

iainmckay, 2015-12-03:
It looks like I has having a derp moment. Both the Servo and analogWrite method allow me to control each ESC at different rates.

Sorry for the inconvenience but thank you very much for the assistance.

closed 2015-12-03 by iainmckay